RoboSense-LiDAR/rslidar_sdk #3: Output of /ros2 topic hz /rslidar_points is NOT stable around 10HZ

BenjaminChi007, 2020-09-07:
Hi Tom,

I am evaluating RoboSense 16 ray Lidar on ROS2 (eloquent/foxy) platform with rslidar_sdk, rslidar_msg and rs_driver. After following the instruction from the github README, the /rslidar_points could be displayed on ROS2 Rviz2, however, topic hz /rslidar_points is NOT stable around 10HZ as the RoboSense package did on ROS1 melodic platform.

Could you give a hand on this problem?  Thanks.

My evaluation work as below:
1. On ROS2 Eloquent and Foxy:
terminal #1: ros2 run rslidar_sdk rslidar_sdk_node
terminal #2: ros2 topic hz /rslidar_points
output of #2:
average rate: 2.356
        min: 0.099s max: 0.801s std dev: 0.32720s window: 4
average rate: 2.224
        min: 0.099s max: 0.801s std dev: 0.26955s window: 6
average rate: 3.334
        min: 0.096s max: 0.801s std dev: 0.23878s window: 13
average rate: 3.265
        min: 0.096s max: 0.801s std dev: 0.22209s window: 16
average rate: 4.001
        min: 0.096s max: 0.801s std dev: 0.20008s window: 24
average rate: 4.133
        min: 0.096s max: 0.801s std dev: 0.19484s window: 31
average rate: 3.933
        min: 0.096s max: 0.801s std dev: 0.19476s window: 35
average rate: 4.001
        min: 0.096s max: 0.801s std dev: 0.18703s window: 40
average rate: 4.273
        min: 0.096s max: 0.801s std dev: 0.17906s window: 47
average rate: 4.499
        min: 0.096s max: 0.801s std dev: 0.17184s window: 54
2. On ROS1 melodic:
terminal #1: roslaunch rslidar_pointcloud rs_lidar_16.launch 
terminal #2: rostopic hz /rslidar_points
subscribed to [/rslidar_points]
average rate: 10.088
        min: 0.092s max: 0.104s std dev: 0.00395s window: 9
average rate: 10.025
        min: 0.092s max: 0.104s std dev: 0.00298s window: 19
average rate: 10.020
        min: 0.092s max: 0.104s std dev: 0.00243s window: 29
average rate: 10.018
        min: 0.092s max: 0.104s std dev: 0.00220s window: 40
average rate: 10.011
        min: 0.092s max: 0.104s std dev: 0.00206s window: 50
^Caverage rate: 10.011
        min: 0.092s max: 0.104s std dev: 0.00203s window: 53

CrazyTomX, 2020-09-08:
Hi Ben, 
   I have not met this kind of problem in our ROS2 environment. Could you add a print function in the following part and we can see whether the original point cloud from decoder is on 10hz.

rslidar_sdk/src/adapter/driver_adapter.hpp

[image]

BenjaminChi007, 2020-09-08:
Hi Tom,

Thanks for instant reply.
I did work as your guide with tiny modification as following:
[image]
According to log dump, it seems driver adapter decoder work all right.
[image]
However, in another terminal, via ros2 topic hz /rslidar_points, the topic hz is around 4hz not 10hz.
[image]

Any further help and guide will be grateful. Thanks.

CrazyTomX, 2020-09-08:
Hi Ben
  From the print message we can see the the decoder has no problem. Since it's should be 10hz and the time interval is 100ms, it means the decoder is correct. Then the problem may be at the ROS2 sender. Could you add a print function in the following part: 

rslidar_sdk/src/adapter/point_cloud_ros_adapter.hpp

[image]

Remember to add the line on the ROS2 part (which is at the bottom of this file).

The sendPointCloud function is registered to driver_adapter as the callback function, once the point cloud is ready, this function will be called and send the point cloud to ROS2. Since we have printed the time in the driver adapter to prove that the frequency of calling this sendPointCloud function is 10hz, then the problem should be in this function.

BenjaminChi007, 2020-09-08:
Hi Tom,

Thanks for your further guide.

Follow your guide, I modified rslidar_sdk/src/adapter/point_cloud_ros_adapter.hpp as following:
[image]
With help of new log, it shows that pub function seems all right as well?
[image]
But, ros2 topic hz /rslidar_points still get wrong hz pulse :(
[image]

CrazyTomX, 2020-09-08:
Hi Ben
  From the screen log info we can see that the pub function is called on 10hz, so I guess the problem may due to the ROS2 itself. I found someone ask for a same question:

https://github.com/ros2/ros1_bridge/issues/218

You can check this out. 

And then I hope you to check the frequency of lidar packets. Set send_packets_ros to true and check the frequency of packets. Since the size of packets is much smaller than point cloud, and the code to publish packets and point cloud are the same. If the packets frequency is 10hz, then the problem should be the ROS2 bandwidth. If the packets frequency is still not 10hz, I will double check the ROS2 publisher code in rslidar_sdk.

CrazyTomX, 2020-09-08:
Note: The rslidar_packets msg is the self-defined msg, so remember to source install/setup.bash in the rslidar_sdk workspace before execute the ros2 topic hz command.

CrazyTomX, 2020-09-08:
BTW, you can also set the parameter start_angle and end_angle to reduce the point cloud size and see whether the frequency can be increased.

BenjaminChi007, 2020-09-08:
> Hi Ben
> From the screen log info we can see that the pub function is called on 10hz, so I guess the problem may due to the ROS2 itself. I found someone ask for a same question:
> 
> [ros2/ros1_bridge#218](https://github.com/ros2/ros1_bridge/issues/218)
> 
> You can check this out.
> 
> And then I hope you to check the frequency of lidar packets. Set send_packets_ros to true and check the frequency of packets. Since the size of packets is much smaller than point cloud, and the code to publish packets and point cloud are the same. If the packets frequency is 10hz, then the problem should be the ROS2 bandwidth. If the packets frequency is still not 10hz, I will double check the ROS2 publisher code in rslidar_sdk.

Thanks for your sharing. packets frequency is stable around 10hz after being enabled.

BenjaminChi007, 2020-09-08:
> BTW, you can also set the parameter start_angle and end_angle to reduce the point cloud size and see whether the frequency can be increased.

After fix start_angle and end_angle respect to 0 and 1, hz of point cloud is still around 4~5 hz.

BenjaminChi007, 2020-09-08:
The platform being used on my hand is Nvidia Xavier as well. I am sorry to see ROS2 latest distributions (foxy and eloquent) are still with point cloud drop issue. I will try to adjust QoS profile setting to check it will help improve point cloud hz dropping.
Thanks for your help. Appreciated.

BenjaminChi007, 2020-09-08:
> Note: The rslidar_packets msg is the self-defined msg, so remember to source install/setup.bash in the rslidar_sdk workspace before execute the ros2 topic hz command.

closed 2020-09-08 by BenjaminChi007