RoboSense-LiDAR/rslidar_sdk #6: Crash on start: Floating point exception in rslidar_sdk_node with RS-LiDAR-M1

duncanmcbryde, 2020-10-07:
Hardware Unit:  Robosense RS-LiDAR-M1
Environment: Ubuntu 20.04, ROS 1 Noetic
Version: RSLidar_SDK Version: v1.2.1, git commit 73b4c66.

Issue: rslidar_sdk_node crashes on reciept of data from lidar unit.

Console output:

```
duncan@rosvm:~/catkin_ws$ roslaunch rslidar_sdk start.launch
... logging to /home/duncan/.ros/log/9b2ce678-08b1-11eb-90d4-5b195771cb2a/roslaunch-rosvm-2308.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://rosvm:38683/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES
  /
    rslidar_sdk_node (rslidar_sdk/rslidar_sdk_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [2316]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9b2ce678-08b1-11eb-90d4-5b195771cb2a
process[rosout-1]: started with pid [2326]
started core service [/rosout]
process[rslidar_sdk_node-2]: started with pid [2329]
process[rviz-3]: started with pid [2334]
********************************************************
**********                                    **********
**********    RSLidar_SDK Version: v1.2.1     **********
**********                                    **********
********************************************************
------------------------------------------------------
Receive Packets From : Online LiDAR
Msop Port: 6699
Difop Port: 7788
------------------------------------------------------
RoboSense-LiDAR-Driver is running.....
[rslidar_sdk_node-2] process has died [pid 2329, exit code -8, cmd /home/duncan/catkin_ws/devel/lib/rslidar_sdk/rslidar_sdk_node __name:=rslidar_sdk_node __log:=/home/duncan/.ros/log/9b2ce678-08b1-11eb-90d4-5b195771cb2a/rslidar_sdk_node-2.log].
log file: /home/duncan/.ros/log/9b2ce678-08b1-11eb-90d4-5b195771cb2a/rslidar_sdk_node-2*.log
[rviz-3] process has finished cleanly
log file: /home/duncan/.ros/log/9b2ce678-08b1-11eb-90d4-5b195771cb2a/rviz-3*.log
```

However there is no file `rslidar_sdk_node-2*.log`.
Running the command directly shows a floating point error, but only once the Lidar has started to send data.

```
duncan@rosvm:~/catkin_ws$ /home/duncan/catkin_ws/devel/lib/rslidar_sdk/rslidar_sdk_node
********************************************************
**********                                    **********
**********    RSLidar_SDK Version: v1.2.1     **********
**********                                    **********
********************************************************
------------------------------------------------------
Receive Packets From : Online LiDAR
Msop Port: 6699
Difop Port: 7788
------------------------------------------------------
RoboSense-LiDAR-Driver is running.....
Floating point exception (core dumped)
```

Which looks like an arithmetic error in rslidar_sdk_node, or one of the libraries.

Wireshark shows valid UDP data streamed from port 6699 from the Lidar unit continuing after node crashes.

[image]

CrazyTomX, 2020-10-07:
Hi, the rslidar_sdk is not support RS-M1 yet. Please use the ros_lidar driver for M1.

duncanmcbryde, 2020-10-08:
Thanks for the clarification. Could you add the list of supported models to README.md, I don't think that information is listed anywhere.

> Please use the ros_lidar driver for M1.

Do you mean [ros_rslidar](https://github.com/RoboSense-LiDAR/ros_rslidar) or [rs_driver](https://github.com/RoboSense-LiDAR/rs_driver)?

duncanmcbryde, 2020-10-08:
[ros_rslidar](https://github.com/RoboSense-LiDAR/ros_rslidar) does not support the M1, according to [issue #47](https://github.com/RoboSense-LiDAR/ros_rslidar/issues/47). I have (patched ros_lidar)[https://github.com/RoboSense-LiDAR/ros_rslidar/issues/49] to make it build successfully, and ros_rslidar can't visualise the output in rviz.

[rs_driver](https://github.com/RoboSense-LiDAR/rs_driver) is a sub-module of [rslidar_sdk](https://github.com/RoboSense-LiDAR/rslidar_sdk), so which does not support the M1.

So am I right in concluding you don't have functionality for the RS-LiDAR-M1 model?

CrazyTomX, 2020-10-08:
I'm sorry for bringing you some unconvience. I've checked with my colleague and the public ros_lidar( git hub version) does not support RS-M1. Now the driver of RS-M1 is not public yet. You can connect the tech-support or sales to get the RS-M1 driver.

duncanmcbryde, 2020-10-12:
I have contacted robosense who have sent a previous version of the rslidar package. However that still requires patching in the same way as https://github.com/RoboSense-LiDAR/ros_rslidar/issues/49, but now times out even though data is recieved.

Can you incorperate the M1 packet structure into this package?

IkeaMu, 2020-10-13:
Hi, Please contact me with email armin.mu@robosense.cn , and mark that you are the person from the github repo. I will help to check this problem.

closed 2020-10-26 by CrazyTomX