RoboSense-LiDAR/rslidar_sdk #7: error encountered when try to play pcap file to ros

w634307272, 2020-10-11:
hello, I'm trying to play pcap file to ros,but no lidar data was displayed in ros,  ' ErrCode_NoDifopRecv' in terminal

CrazyTomX, 2020-10-11:
Hi, the default setting of the driver is that it must receive the difop packet to decode the data. So please check out whether your pcap file contains difop packet, and make sure the port number of difop is correct.  If you want to decode data without difop, you can set the hiding parameter wait_for_difop to false( refer the hiding parameter document). In this way, you can decode the data without difop, but the output pointcloud may be uncorrect(may show as a 2D plane).

w634307272, 2020-10-11:
> Hi, the default setting of the driver is that it must receive the difop packet to decode the data. So please check out whether your pcap file contains difop packet, and make sure the port number of [...]

I'm not sure whether this pcap file have difop,we record this pcap with robosense's offical software yesterday, my computer hasn't connected to RS80 yet, the file was recorded by my classmate's computer and he sent the file to me. And I tried to decode without difop, only show as a 2D plane and 'ErrCode_WrongPktHeader' in Terminal

CrazyTomX, 2020-10-11:
Please use WireShark( a third party tool) to record pcap file. The RsView will not record difop.

w634307272, 2020-10-11:
Ah... thank you! my last question, I want to use this pcap as an input to lego-loam to build a map. but I found that everytime I input 'roslaunch  rslidar_sdk start.launch' , It will start a new rviz and close lego-loam's rviz. Do you know how can I solve this problem? thx!

CrazyTomX, 2020-10-11:
You can modify the launch file, comment or just delete the rviz part. Then the driver will no more start a new rviz.

w634307272, 2020-10-11:
> You can modify the launch file, comment or just delete the rviz part. Then the driver will no more start a new rviz.

got it, thank you!

closed 2020-10-11 by w634307272