RoboSense-LiDAR/rslidar_sdk #9: Add Ring Field

TillBeemelmanns, 2020-11-03:
Hi, 

thanks for the new sdk. I am using your kit along with the RS32, however I am missing the ring information field for the point cloud when I am using the "rslidar_sdk_node". I am using the "offline mode" where my packet and difop messages come from a ROS bag. I know that the sdk can decode the ring field but how can I add that field ?

Best Regards

CrazyTomX, 2020-11-04:
Sorry that for the **rslidar_sdk v1.2.1**, its not support to decode the ring field of LiDAR. But the driver core **rs_driver v1.2.1** does support the **ring** and **timestamp** of each point. Which means you need to do some advanced development with the **rs_driver** to publish the pointcloud with ring field to ROS.

BTW, we are going to release the v1.3.0 around Nov.15th. For **rslidar_sdk v1.3.0**, it does support sending pointcloud with ring field and timestamp for each point to ROS. 

If you have some urgent requirement, please left your email address here and I will sent you the alpah-test version of v1.3.0. You may use it for now and update to release version later.

TillBeemelmanns, 2020-11-04:
Thanks for the quick response. I will wait for the new release :)

closed 2020-11-04 by TillBeemelmanns