RoboSense-LiDAR/rslidar_sdk #10: Timestamping of the Pointcloud in ROS

Patrick-Lascombe, 2020-11-18:
Hello, 
With the paramerter use_lidar_clock = true, the timestamping of the ROS sensor_msgs::PointCloud2 message. The call to get the lidar clock is executed on the first packed of the scan or the last packet of the scan ?

Thank you for your help

CrazyTomX, 2020-11-19:
Hi,
The timestamp of the point cloud is the time of the **last** packet of the scan.

Patrick-Lascombe, 2020-11-19:
Thanks, and also thank you for the 1.3.0 release the ring and timestamp info are really valuables !

closed 2020-11-19 by Patrick-Lascombe

RonaldEnsing, 2024-11-28:
Note that other sensors (such as Ouster) use the **first** packet of the scan, while the RoboSense is using the **last** packet of the scan for time stamping the point cloud. This can give some confusion. Autoware specifies the following:

> The header of the point cloud message is expected to have the time of the earliest point it has.