RoboSense-LiDAR/rslidar_sdk #13: "ERRCODE_WRONGPKTHEADER"

SolarSLAM, 2020-11-24:
After I compiled this package,  it didn't work

=================== =================================================================
roslaunch rslidar_sdk start.launch                         20-11-25 - 0:29:43
... logging to /home/nvidia/.ros/log/a8e42758-2e73-11eb-b1df-48b02d0e3d9f/roslaunch-tegra-ubuntu-13507.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://tegra-ubuntu:34452/
SUMMARY
PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.16
NODES
  /
  rslidar_sdk_node (rslidar_sdk/rslidar_sdk_node)
 viz (rviz/rviz)
auto-starting new master
process[master]: started with pid [13517]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a8e42758-2e73-11eb-b1df-48b02d0e3d9f
process[rosout-1]: started with pid [13530]
started core service [/rosout]
process[rslidar_sdk_node-2]: started with pid [13541]
  RSLidar_SDK Version: v1.3.0 
process[rviz-3]: started with pid [13552]
Receive Packets From : Online LiDAR
Msop Port: 6699
Difop Port: 7788
Send PointCloud To : ROS
PointCloud Topic: /rslidar_points
RoboSense-LiDAR-Driver is running.....
ERRCODE_WRONGPKTHEADER
ERRCODE_WRONGPKTHEADER
ERRCODE_WRONGPKTHEADER
ERRCODE_WRONGPKTHEADER
ERRCODE_WRONGPKTHEADER
ERRCODE_WRONGPKTHEADER
ERRCODE_WRONGPKTHEADER
ERRCODE_WRONGPKTHEADER
ERRCODE_WRONGPKTHEADER
ERRCODE_WRONGPKTHEADER
ERRCODE_WRONGPKTHEADER

CrazyTomX, 2020-11-25:
Hi,
  What's your LiDAR type? Please make sure the lidar_type argument is set up correctly in config.yaml.

SolarSLAM, 2020-11-26:
OOh, I didn't notice the config.yaml . It works .Thanks for your help

closed 2020-11-26 by SolarSLAM