RoboSense-LiDAR/rslidar_sdk #16: dense pointcloud

yzzymt, 2020-12-29:
Hi,
Can we get dense pointcloud directly from 1.3 sdk by changing some config parameters?
When I work with rslidar 16, I need remove NaN from the pointcloud first.

HaoQChen, 2020-12-29:
according to `pcl/point_cloud.h`

`is_dense - specifies if all the data in points is finite (true), or whether it might contain Inf/NaN values (false). ` 

when some condition occur, point will be set to NAN in `driver/decoder/decoder_RS16.hpp/decodeMsopPkt`:

[image]

maybe you can change NAN to 0.0001 and set `pcl_pointcloud->is_dense` to true in your `pointCloudCallback`

yzzymt, 2020-12-29:
I see, thank you so much for your quick response

closed 2020-12-29 by yzzymt