RoboSense-LiDAR/rslidar_sdk #18: Feature Request: Account for ego motion during one rotation of the sensor

TillBeemelmanns, 2021-02-17:
The rolling shutter effect of a rotating sensor is well known. The ego motion may induce some undesirable artifacts for some applications. For example, sensor fusion of LiDAR and camera. Hence, I would like to request an implementation similar to the [Velodyne driver](https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/src/conversions/transform.cc#L135), where each packet of a rotation can be transformed to a fixed frame to account for the ego motion.

HaoQChen, 2021-02-18:
You mean provide time stamp for each point? This has been implemented, you can add a `timestamp` member to your pointcloud class

ego motion need motion data like IMU, or you mean provide an interface for user to add there motion data?

TillBeemelmanns, 2021-02-18:
I don't mean the timestamp. 

I am considering the case when the driver is used by ROS. The ego motion can then be provided by the ROS TF2 Transform package, by setting up a `tf2_ros::TransformListener` which buffers the transformations that occurred during one revolution of the LiDAR. These transforms can then be used to transform the points of each package onto single fixed frame at specific timestamp (e.g. the timestamp of the last package).