RoboSense-LiDAR/rslidar_sdk #19: rs-16 and rs-m1 supporting problem

simulacrus, 2021-02-18:
0. Configuration rslidar_sdk

  ```yaml
common:
  msg_source: 1                                         #0: not use Lidar
                                                        #1: packet message comes from online Lidar
                                                        #2: packet message comes from ROS or ROS2
                                                        #3: packet message comes from Pcap file
                                                        #4: packet message comes from Protobuf-UDP
                                                        #5: point cloud comes from Protobuf-UDP
  send_packet_ros: false                                #true: Send packets through ROS or ROS2(Used to record packet)
  send_point_cloud_ros: true                            #true: Send point cloud through ROS or ROS2
  send_packet_proto: false                              #true: Send packets through Protobuf-UDP
  send_point_cloud_proto: false                         #true: Send point cloud through Protobuf-UDP
  pcap_path: /home/robosense/lidar.pcap                 #The path of pcap file

lidar:

#top lidr info

  - driver:
      lidar_type: RS32             #LiDAR type - RS16, RS32, RSBP, RS128, RS80, RSM1, RSHELIOS
      frame_id: /lidar_top         #Frame id of message
      msop_port: 6690              #Msop port of lidar
      difop_port: 7789             #Difop port of lidar
      start_angle: 0               #Start angle of point cloud
      end_angle: 360               #End angle of point cloud 
      min_distance: 0.2            #Minimum distance of point cloud
      max_distance: 200            #Maximum distance of point cloud
      use_lidar_clock: false       #True--Use the lidar clock as the message timestamp
                                   #False-- Use the system clock as the timestamp  
    ros:
      ros_recv_packet_topic: /rslidar_top_packets          #Topic used to receive lidar packets from ROS
      ros_send_packet_topic: /rslidar_top_packets          #Topic used to send lidar packets through ROS
      ros_send_point_cloud_topic: /lidar_top/points      #Topic used to send point cloud through ROS
    proto:
      point_cloud_recv_port: 60021                     #Port number used for receiving point cloud 
      point_cloud_send_port: 60021                     #Port number which the point cloud will be send to
      msop_recv_port: 60022                            #Port number used for receiving lidar msop packets
      msop_send_port: 60022                            #Port number which the msop packets will be send to 
      difop_recv_port: 60023                           #Port number used for receiving lidar difop packets
      difop_send_port: 60023                           #Port number which the difop packets will be send to 
      point_cloud_send_ip: 127.0.0.1                   #Ip address which the point cloud will be send to 
      packet_send_ip: 127.0.0.1                        #Ip address which the lidar packets will be send to
  
# left lidar info

  - driver:
      lidar_type: RS16             #LiDAR type - RS16, RS32, RSBP, RS128, RS80, RSM1, RSHELIOS
      frame_id: /lidar_left         #Frame id of message
      msop_port: 5590              #Msop port of lidar
      difop_port: 6689             #Difop port of lidar
      start_angle: 0               #Start angle of point cloud
      end_angle: 360               #End angle of point cloud 
      min_distance: 0.2            #Minimum distance of point cloud
      max_distance: 200            #Maximum distance of point cloud
      use_lidar_clock: false       #True--Use the lidar clock as the message timestamp
                                   #False-- Use the system clock as the timestamp  
    ros:
      ros_recv_packet_topic: /rslidar_left_packets          #Topic used to receive lidar packets from ROS
      ros_send_packet_topic: /rslidar_left_packets          #Topic used to send lidar packets through ROS
      ros_send_point_cloud_topic: /lidar_left/points      #Topic used to send point cloud through ROS
    proto:
      point_cloud_recv_port: 60021                     #Port number used for receiving point cloud 
      point_cloud_send_port: 60021                     #Port number which the point cloud will be send to
      msop_recv_port: 60022                            #Port number used for receiving lidar msop packets
      msop_send_port: 60022                            #Port number which the msop packets will be send to 
      difop_recv_port: 60023                           #Port number used for receiving lidar difop packets
      difop_send_port: 60023                           #Port number which the difop packets will be send to 
      point_cloud_send_ip: 127.0.0.1                   #Ip address which the point cloud will be send to 
[... 60 lines cut ...]

After `roslaunch rslidar_sdk start.launch` and opening rviz i was confusing by periodically disappearing part of points.  
(View 1.pcd, 2.pcd, 1.png, 2.png).

[1.pcd](https://github.com/RoboSense-LiDAR/rslidar_sdk/files/6000995/1.zip)

[image]

[2.pcd](https://github.com/RoboSense-LiDAR/rslidar_sdk/files/6000996/2.zip)

[image]

Interesting that RSView show scans correctly

2.  # RS m1

There is some problem with correcting reading poins from RS MEMS by rslidar_sdk

[3.pcd](https://github.com/RoboSense-LiDAR/rslidar_sdk/files/6000999/3.zip)

[image]

RSView show correcting scans

[image]

HaoQChen, 2021-02-19:
1. **periodically disappearing part of points**

did you change `split_frame_mode` to 2 or 3?

[image]

2. which version of RSView did you use? There are two modes to parse points, And this driver only support one mode. You need to content our support colleague, he will help you about it

armin.mu@robosense.cn

HaoQChen, 2021-03-04:
Can this issue be closed?

simulacrus, 2021-03-15:
I'm trying to configure rslidar_sdk with split_frame_mode =2. Result ...
[rs16.zip](https://github.com/RoboSense-LiDAR/rslidar_sdk/files/6140268/rs16.zip)

Lidar scan looks same with spilit_frame_mode = 3 and num_pkts_split = 37. If I set num_pkts_split > 38 then i have oversize pointcloud

HaoQChen, 2021-03-15:
> I'm trying to configure rslidar_sdk with split_frame_mode =2. Result ...
> [rs16.zip](https://github.com/RoboSense-LiDAR/rslidar_sdk/files/6140268/rs16.zip)
> 
> Lidar scan looks same with spilit_frame_mode = 3 and num_pkts_split = 37. If I set num_pkts_split > 38 then i have oversize pointcloud

emm, I mean you need to set split_frame_mode to 1, default is 1.

If split frame by cut angle, driver will split frames when it finds lidar step over cut_angle. This guarantees no periodically disappearing part of points.

If split frame by number of packets, unstable speed will lead to a  periodically disappearing part of points

simulacrus, 2021-03-15:
> emm, I mean you need to set split_frame_mode to 1, default is 1.
>
> If split frame by cut angle, driver will split frames when it finds lidar step over cut_angle. This guarantees no periodically [...]

Initially, I used the default parameters. 
I also debugged the driver and saw that scans are periodically compiled from 37 to 38 packages. Maybe this is the problem

Andrea2202, 2021-05-20:
I've the same problem with ROS2 and M1.
Have you any solutions?

HaoQChen, 2021-06-01:
> I've the same problem with ROS2 and M1.
> Have you any solutions?
> 
> @HaoQChen

Did you try split_frame_mode? 

Did you try ROS1?

The environment of your computer?

tomcamp0228, 2021-06-02:
> > I've the same problem with ROS2 and M1.
> > Have you any solutions?
> > @HaoQChen
> 
> Did you try split_frame_mode?
> 
> Did you try ROS1?
> 
> The environment of your computer?

Hello, I have same problem with RS-M1-B2 as @simulacrus shows. I'm using about the latest version of your SDK and the enviorment is Ubuntu 18.04 with ROS eloquent. How can I solve the problem? Waiting for your reply, thank you.

HaoQChen, 2021-06-10:
> Hello, I have same problem with RS-M1-B2 as @simulacrus shows. I'm using about the latest version of your SDK and the enviorment is Ubuntu 18.04 with ROS eloquent. How can I solve the problem? [...]

`rs_driver` does not support B2, you need to use `ros_rslidar`