RoboSense-LiDAR/rslidar_sdk #20: RS16 point cloud looks flatten

esaumar, 2021-03-10:
These are the version and settings I'm using:
**Ubuntu: 18.04
Ros: Melodic
Lidar: RS16
Point type: XYZIRT and XYZI**

I followed all steps to install the sdk with ROS (Section 4.2 from the readme) but the point cloud I watch in Rviz looks flat. 
[image]
[image]

Besides, the values from the PointCloud2 message are **always zero** for Ring and Z
[image]

When I use the **previous version** of the driver:
https://github.com/RoboSense-LiDAR/ros_rslidar
Rviz shows the point cloud correctly.
[image]

This is the **config.yaml** file I am using
[image]

Do you know what might be the cause of this issue?
Any help would be appreciated.
Thanks!

HaoQChen, 2021-03-15:
1. Check your difop port, if it is wrong, vertical angle may miss when decoding data

2. If you are using a very old version RS16, angle message will not sent out by difop, you need to load it manually

esaumar, 2021-03-19:
Thanks for the answer @HaoQChen. It seems that I have an old version of RS16. I loaded it manually and now I have the full point cloud.
In the `config.yaml` file, I added a line `angle_path` under `driver` like this:

```
lidar:
  - driver:
      lidar_type: RS16             #LiDAR type - RS16, RS32, RSBP, RS128, RS80, RSM1, RSHELIOS
      frame_id: /rslidar           #Frame id of message
      msop_port: 6699              #Msop port of lidar
      difop_port: 7788             #Difop port of lidar
      start_angle: 0               #Start angle of point cloud
      end_angle: 360               #End angle of point cloud 
      min_distance: 0.2            #Minimum distance of point cloud
      max_distance: 200            #Maximum distance of point cloud
      use_lidar_clock: false       #True--Use the lidar clock as the message timestamp
                                   #False-- Use the system clock as the timestamp  
      angle_path: /home/user/angle.csv

```
I copied the file `angle.csv` from https://github.com/RoboSense-LiDAR/ros_rslidar/tree/master/rslidar_pointcloud/data/rs_lidar_16
and I added a column of zeros because it seems that it is the format the difop port publishes the data.

```
-14.9994,0
 -12.9766,0
 -10.9894,0
 -8.9750,0
 -6.9734,0
 -4.9900,0
 -2.9946,0
 -0.9918,0
 14.9860,0
 12.9630,0
 10.9756,0
 8.9960,0
 6.9945,0
 5.0113,0
 2.9803,0
 1.0133,0
```
Thanks for the help!

closed 2021-03-19 by esaumar

HaoQChen, 2021-03-20:
yes, second column is horizontal angle, and all lasers of RS16 is in one column, so it is 0（RS32 is different）

Old version, we will give you a USB disk, there is a angle file in it. This one will more fit your lidar