RoboSense-LiDAR/rslidar_sdk #21: ERRCODE_MSOPTIMEOUT

VimalMollyn, 2021-03-12:
Hi, 

I'm trying to set up an RS Helios lidar. I updated the config.yaml file as follows:

```
common:
  msg_source: 1                                         #0: not use Lidar
                                                        #1: packet message comes from online Lidar
                                                        #2: packet message comes from ROS or ROS2
                                                        #3: packet message comes from Pcap file
                                                        #4: packet message comes from Protobuf-UDP
                                                        #5: point cloud comes from Protobuf-UDP
  send_packet_ros: false                                #true: Send packets through ROS or ROS2(Used to record packet)
  send_point_cloud_ros: true                            #true: Send point cloud through ROS or ROS2
  send_packet_proto: false                              #true: Send packets through Protobuf-UDP
  send_point_cloud_proto: false                         #true: Send point cloud through Protobuf-UDP
  pcap_path: /home/robosense/lidar.pcap                 #The path of pcap file

lidar:
  - driver:
      lidar_type: RSHELIOS            #LiDAR type - RS16, RS32, RSBP, RS128, RS80, RSM1, RSHELIOS
      frame_id: /rslidar           #Frame id of message
      msop_port: 6699              #Msop port of lidar
      difop_port: 7788             #Difop port of lidar
      start_angle: 0               #Start angle of point cloud
      end_angle: 360               #End angle of point cloud 
      min_distance: 0.2            #Minimum distance of point cloud
      max_distance: 200            #Maximum distance of point cloud
      use_lidar_clock: false       #True--Use the lidar clock as the message timestamp
                                   #False-- Use the system clock as the timestamp  
    ros:
      ros_recv_packet_topic: /rslidar_packets          #Topic used to receive lidar packets from ROS
      ros_send_packet_topic: /rslidar_packets          #Topic used to send lidar packets through ROS
      ros_send_point_cloud_topic: /rslidar_points      #Topic used to send point cloud through ROS
    proto:
      point_cloud_recv_port: 60021                     #Port number used for receiving point cloud 
      point_cloud_send_port: 60021                     #Port number which the point cloud will be send to
      msop_recv_port: 60022                            #Port number used for receiving lidar msop packets
      msop_send_port: 60022                            #Port number which the msop packets will be send to 
      difop_recv_port: 60023                           #Port number used for receiving lidar difop packets
      difop_send_port: 60023                           #Port number which the difop packets will be send to 
      point_cloud_send_ip: 127.0.0.1                   #Ip address which the point cloud will be send to 
      packet_send_ip: 127.0.0.1                        #Ip address which the lidar packets will be send to
```

**Still getting the error shown above.**
This is what my Helios config looks like:

[image]

Any help would be appreciated.

Thanks!
Vimal

HaoQChen, 2021-03-15:
Could you record a pcap file to us?

closed 2021-04-08 by HaoQChen

wangshichang-726, 2022-07-20:
I have the same problem, how did you solve it?

TZECHIN6, 2022-11-04:
I face the same issue as well. How do you solve that?

wangshichang-726, 2022-11-04:
> 

maybe the IP issue

TZECHIN6, 2022-11-04:
For me its not a IP issue, I have check everything and message can be see in Wireshark. The problem is caused by my flooded switch, which I accidentally connected multiple velodyne lidar before, and they broadcast udp message which flooded the device somehow. Power off the switch and everythings just work fine.

I don't know what is the root cause, even I directly connect to the lidar with host still got this issue. Is is really caused by the flooded switch is hard to tell. but it works somehow.

TZECHIN6, 2023-02-28:
For the web UI, just like working with velodrome, assuming you haven't set the IP of RS LiDAR before, make sure your PC is also in a same subnet, then enter the default IP in the browser, you will enter the webUI.

arwinder-jaspal, 2023-03-01:
> For the web UI, just like working with velodyne, assuming you haven't set the IP of RS LiDAR before, make sure your PC is also in a same subnet, then enter the default IP in the browser, you will enter the webUI.

Unable to access UI.

TZECHIN6, 2023-03-01:
What UI are you mentioning? Would you provide more information, otherwise others hard to help. Thanks

alphonsustay, 2023-03-24:
@arwinder-jaspal @TZECHIN6  , hope this might still be helpful for you guys.

My Environment: Ubuntu 20.04 with ROS2 Galactic

I was able to get the sensor readings after setting the Lidar's destination IP address (Under the Settings Tab) to my PC's IP address. The default destination is 192.168.1.102 . An alternative would be to set your PC's IP address to the sensor's default destination IP.

The robot settings can be access via any web browser, the Lidar's default IP address should be 192.168.1.200. As long as you set your PC's ethernet IP address within the same subnet (255.255.255.0), you should be able to access the 'settings' page when you enter the lidar's IP address in your web browser.

TZECHIN6, 2024-04-26:
> Has anyone solved this issue? I am on Ubuntu 22.04 with ROS 2 Humble. With a 1sec interval I receive an error message: ERRCODE_MSOPTIMEOUT
> 
> Wireshark shows DIFOP and MSOP packages.

1. Make sure you can ping the LiDAR (same subnet)
2. Confirm the ROS2 driver config setting is right.
3. If nothing works, try connect it without switch.

maxpolzin, 2024-04-26:
Thank you for your prompt response.

1. Pinging is no problem.
2. I added an image of the Ros driver settings. Am I missing sth. in the settings?
3. That's a bit tricky atm, because of our network setup.

maxpolzin, 2024-04-29:
@TZECHIN6 Thanks again for your response. 

I connected the BPearl directly to our computer via an Ethernet dongle. The error remains the same.
I uploaded our configuration parameters here: https://github.com/RoboSense-LiDAR/rslidar_sdk/issues/139

The configuration is basically a fresh clone of the repo with the only change being in the config file to RSBP (for BPearl) on a clean Ubuntu 22.04 installation. Wireshark indicates MSOP and DIFOP packages.

Do you have any ideas what could go wrong? Does our BPearl maybe have an incompatible firmware installed?

maxpolzin, 2024-04-29:
I am able to capture packages with Wireshark into a pcap file and visualize them with the RSViewer GUI tool. Uploaded screenshots of our configuration are here: https://github.com/RoboSense-LiDAR/rslidar_sdk/issues/139

Sorry for the duplicated issue. If they are related, feel free to close one.

nicksemin, 2025-02-02:
If anyone still experiences this, double check your config file. I noticed that I had the wrong lidar type set, hence, that issue

robbie2021, 2025-03-04:
> g

> If anyone still experiences this, double check your config file. I noticed that I had the wrong lidar type set, hence, that issue

hi， i face the problem, but i can show thw lidar cloud through pcap, so maybe it's not the lidar type problem?

eegmnn, 2025-09-18:
I had the same issue. I noticed that, in the web UI, which you can access by typing your device ip to your browser, target IP should match your computers IP address.