RoboSense-LiDAR/rslidar_sdk #24: rviz flat pointcloud + message built error

WillAarents, 2021-03-24:
Hey everyone, 

we are using the Robosense Ruby (RS128) and want to publish and record the data via ROS (Ubuntu 18.04 and Melodic). However, after setting all necessary configs in the config.yaml and starting `roslaunch rslidar_sdk start.launch`, we receive the following rviz output:
[image]
[image]

We also checked the published topics via `rostopic list`, and all are visible. (/rslidar_packets        /rslidar_packets_difop  /rslidar_points) 
When we use `rostopic echo`, only /rslidar_points are visible, whereas /rslidar_packets and /rslidar_packets_difop give the following output:
`ERROR: Cannot load message class for [rslidar_msgs/rslidarScan]. Are your messages built?`

We also checked the pointcloud via RSView and that seems to work totally fine.

Do you have any hints/ ideas what could solve our problem?

Thanks.

PS: Attached the config.yaml file content we use:

[
common:
  msg_source: 1                                         #0: not use Lidar
                                                        #1: packet message comes from online Lidar
                                                        #2: packet message comes from ROS or ROS2
                                                        #3: packet message comes from Pcap file
                                                        #4: packet message comes from Protobuf-UDP
                                                        #5: point cloud comes from Protobuf-UDP
  send_packet_ros: true                                 #true: Send packets through ROS or ROS2(Used to record packet)
  send_point_cloud_ros: true                            #true: Send point cloud through ROS or ROS2
  send_packet_proto: false                              #true: Send packets through Protobuf-UDP
  send_point_cloud_proto: false                         #true: Send point cloud through Protobuf-UDP
  pcap_path: /home/robosense/lidar.pcap                 #The path of pcap file

lidar:
  - driver:
      lidar_type: RS128            #LiDAR type - RS16, RS32, RSBP, RS128, RS80, RSM1, RSHELIOS
      frame_id: /rslidar           #Frame id of message
      msop_port: 6699              #Msop port of lidar
      difop_port: 7788             #Difop port of lidar
      start_angle: 0               #Start angle of point cloud
      end_angle: 360               #End angle of point cloud 
      min_distance: 0.2            #Minimum distance of point cloud
      max_distance: 200            #Maximum distance of point cloud
      use_lidar_clock: true       #True--Use the lidar clock as the message timestamp
                                   #False-- Use the system clock as the timestamp  
    ros:
      ros_recv_packet_topic: /rslidar_packets          #Topic used to receive lidar packets from ROS
      ros_send_packet_topic: /rslidar_packets          #Topic used to send lidar packets through ROS
      ros_send_point_cloud_topic: /rslidar_points      #Topic used to send point cloud through ROS
    proto:
      point_cloud_recv_port: 60021                     #Port number used for receiving point cloud 
      point_cloud_send_port: 60021                     #Port number which the point cloud will be send to
      msop_recv_port: 60022                            #Port number used for receiving lidar msop packets
      msop_send_port: 60022                            #Port number which the msop packets will be send to 
      difop_recv_port: 60023                           #Port number used for receiving lidar difop packets
      difop_send_port: 60023                           #Port number which the difop packets will be send to 
      point_cloud_send_ip: 127.0.0.1                   #Ip address which the point cloud will be send to 
      packet_send_ip: 127.0.0.1                        #Ip address which the lidar packets will be send to
]

WillAarents, 2021-03-29:
[Update]
Our issue regarding the flat pointcloud visualization is related to Issue #20 . We needed the corresponding angle.csv file to correct this issue. 

angle.csv
```bash
-13.565 | 5.95
-- | --
-1.09 | 4.25
-4.39 | 2.55
1.91 | 0.85
-6.65 | 5.95
-0.29 | 4.25
-3.59 | 2.55
2.71 | 0.85
-5.79 | 5.95
0.51 | 4.25
-2.79 | 2.55
3.51 | 0.85
-4.99 | 5.95
1.31 | 4.25
-1.99 | 2.55
5.06 | 0.85
-4.19 | 5.95
2.11 | 4.25
-19.582 | 2.55
-1.29 | 0.85
-3.39 | 5.95
2.91 | 4.25
-7.15 | 2.55
-0.49 | 0.85
-2.59 | 5.95
3.71 | 4.25
-5.99 | 2.55
0.31 | 0.85
-1.79 | 5.95
5.96 | 4.25
-5.19 | 2.55
1.11 | 0.85
-0.99 | 5.95
-4.29 | 4.25
2.01 | 2.55
-25 | 0.85
-0.19 | 5.95
-3.49 | 4.25
2.81 | 2.55
-7.65 | 0.85
0.61 | 5.95
-2.69 | 4.25
3.61 | 2.55
-6.09 | 0.85
1.41 | 5.95
-1.89 | 4.25
5.46 | 2.55
-5.29 | 0.85
2.21 | 5.95
-16.042 | 4.25
-1.19 | 2.55
-4.49 | 0.85
3.01 | 5.95
-6.85 | 4.25
-0.39 | 2.55
-3.69 | 0.85
3.81 | 5.95
-5.89 | 4.25
0.41 | 2.55
-2.89 | 0.85
6.56 | 5.95
-5.09 | 4.25
1.21 | 2.55
-2.09 | 0.85
-8.352 | -0.85
-0.69 | -2.55
-3.99 | -4.25
2.31 | -5.95
-6.19 | -0.85
0.11 | -2.55
-3.19 | -4.25
3.11 | -5.95
-5.39 | -0.85
0.91 | -2.55
-2.39 | -4.25
3.96 | -5.95
-4.59 | -0.85
1.71 | -2.55
-1.59 | -4.25
7.41 | -5.95
-3.79 | -0.85
2.51 | -2.55
-10.346 | -4.25
-0.89 | -5.95
-2.99 | -0.85
3.31 | -2.55
-6.39 | -4.25
-0.09 | -5.95
-2.19 | -0.85
4.41 | -2.55
-5.59 | -4.25
0.71 | -5.95
-1.39 | -0.85
11.5 | -2.55
-4.79 | -4.25
1.51 | -5.95
-0.59 | -0.85
-3.89 | -2.55
2.41 | -4.25
-11.742 | -5.95
0.21 | -0.85
-3.09 | -2.55
3.21 | -4.25
-6.5 | -5.95
1.01 | -0.85
-2.29 | -2.55
4.16 | -4.25
-5.69 | -5.95
1.81 | -0.85
-1.49 | -2.55
9 | -4.25
-4.89 | -5.95
2.61 | -0.85
-9.244 | -2.55
-0.79 | -4.25
-4.09 | -5.95
3.41 | -0.85
-6.29 | -2.55
0.01 | -4.25
-3.29 | -5.95
4.71 | -0.85
-5.49 | -2.55
0.81 | -4.25
-2.49 | -5.95
15 | -0.85
-4.69 | -2.55
1.61 | -4.25
-1.69 | -5.95
```

However, for the error message we have not found a solution yet.

HaoQChen, 2021-03-31:
Hi, as I know, all RS128 will send out DIFOP with angle message. User dont need to load angle.csv by himself.

Could you capture the UDP data with Wireshark or something else, and check if DIFOP port is what you set in config.yaml?

default setting is `difop_port: 7788             #Difop port of lidar` 

you may need wireshark filter: `udp.port!=6699`

closed 2021-04-08 by HaoQChen