RoboSense-LiDAR/rslidar_sdk #25: Error while executing launch file : can't locate node

arizk2605, 2021-03-30:
Hi everyone, 
I cloned the driver's git in the catkin workspace of my raspberry pi 4. I followed the instructions in the readme file but couldn't launch the start_launch.launch file. I keep on getting this error : 

**ERROR: cannot launch node of type [rslidar_sdk/rslidar_sdk_node]: Cannot locate node of type [rslidar_sdk_node] in package [rslidar_sdk]. Make sure file exists in package path and permission is set to executable (chmod +x)**

Even though I run "chmod +x" on the node itself and I successfully compiled the package. Do you have any clue what might be causing this problem ?

Also, i'm working on an acquisition unit and  i'd like to save the raw data of lidar into rosbag automatically using the driver. If you have any quick ideas to do that it'll be helpful too !

HaoQChen, 2021-03-31:
did you install ROS1? which version

you can just run the node with `./rslidar_sdk_node` in your path

you can set `send_point_cloud_ros: true`, the node will publish a ros topic of pointcloud

arizk2605, 2021-03-31:
I'm currently using ROS melodic in a raspberry pi 4.
I tried that too but didn't work..

arizk2605, 2021-03-31:
Here are the errors i get when i use "./rslidar_sdk_node" :

**./rslidar_sdk_node.cpp: line 1: /bin: Is a directory
./rslidar_sdk_node.cpp: line 2: syntax error near unexpected token `c'
./rslidar_sdk_node.cpp: line 2: `Copyright (c) 2020 RoboSense'**

HaoQChen, 2021-03-31:
umm, You don't seem to have compiled the code into an executable program

did you find an executable file with name rslidar_sdk_node?

[image]

if you want to use catkin tool to compile, you need to follow section 4.2 in README

[image]

arizk2605, 2021-03-31:
I followed every step described in the readme, as i said, but it didn't work infortunately.
How did you manage to display what is shown in the screenshot ?

HaoQChen, 2021-04-02:
`ls` command

You have better try that in your pc if you are not familiar with ubuntu or ROS. raspberry pi need some additional step to compile ROS or PCL as I knowed

HaoQChen, 2021-04-02:
`ls -l` with `-l` option will show long list info

arizk2605, 2021-04-08:
Hi again, 
I'm currently running it in Ubuntu, i followed every step of the readme file  but when launching it keeps sending the message message : 
ERRCODE_MSOPTIMEOUT
ERRCODE_DIFOPTIMEOUT

I checked the datasheet and i'm pretty sure the config is right...DO you have any clue about the source of the problem ?

arizk2605, 2021-04-08:
Here is my  config.yaml : 
 
[image]

Here are the error messages I get : 

[image]

arizk2605, 2021-04-08:
Also, i used wireshark to find the correct MSOP DIFOP parameters and display the messages but i only get this : 
[image]

HaoQChen, 2021-04-09:
umm, your lidar is not connect to computer? From your wireshark, not thing captured

Connect lidar to your computer and set your NIC to the same LAN as the lidar

then your wireshark will capture many UDP from lidar, that is what we call MSOP or DIFOP

arizk2605, 2021-04-09:
I connected the lidar to the pc using the RG45 cable. 
What's the NIC and how can I proceed to set it to the LAN ?

HaoQChen, 2021-04-09:
NIC  is net interface card, 

You just connect to computer on physical layer

not Network Layer, computer cannot communicate with lidar

arizk2605, 2021-04-09:
Ok I see,
Do you know how i could do that in ubuntu ?

arizk2605, 2021-04-12:
Hi again, 
I managed to catch some UDP-type of data but i can't seem to find the right ports (MSOP and DIFOP). Do you have any clue on how to use what i found ?
[image]

[image]

HaoQChen, 2021-04-19:
Please contact armin.mu@robosense.cn

arwinder-jaspal, 2023-03-01:
I am having the same error but using wireshark I am able to see there are data being sent via UDP to the correct msop port. But when I view it through RViz no data is shown, rostopic echo returns nothing as well. 

[image]
[image]
[image]

eegmnn, 2024-07-25:
@arwinder-jaspal Have you found a solution on the topic? If so, can you update the issue?

krunalmbhatt, 2025-04-01:
I tried the following things for RSAIRY Lidar. I am using Ubuntu 22.04 and ran the package in a docker container. 
1) Check if you can ping Lidar at 192.168.1.200
2) Check port number and Lidar type in config. 
3) Check if your computer's IP has been changed correspond to Lidar's destination IP (default IP is 192.168.1.102)
4) If you are still not getting the error code, please install Wireshark and check if you can get the UDP data stream normally. If you get udp packets, then try running: sudo dfw allow 6699/udp, sudo dfw allow 7788/udp, on the local system. Then execute the same command inside docker container as well.