RoboSense-LiDAR/rslidar_sdk #27: Wrong returned distances in RS16

mtlazaro, 2021-04-05:
Hi,

We are trying to use the rslidar_sdk package with our RS16 instead of the old ros_rslidar.

We are using the provided "start.launch" in this package. We manage to connect with the RS16 and visualize its data.

However we are observing a problem regarding the range data returned by the sensor, as you can observe from the attached images, the ranges (colored pointcloud) are much shorter than the real distances measured with a 2D laserscan (white points).
In the images we have limited the FOV of the LiDAR but we observe the same problem if we use the full 360 degrees.

When we use the ros_rslidar package the distances are correct.

[image]
[image]

mjseppa, 2021-04-06:
Are you running older lidar without difop, or wrong difop port, so driver assumes it is in 0.5cm resolution instead of 1cm resolution?

mtlazaro, 2021-04-06:
Hi,
we acquired our RS16 on January 2019.
We are not doing anything special when using the start.launch file.
We have just limited the FOV in the config.yaml but the rest of the parameters have the predefined values.

I attach a screenshot of our terminal and our config.yaml.

[image]

[image]

mjseppa, 2021-04-06:
Looking at the driver it does not seem to support 1cm resolution lidars?
in rs_driver there is "constexpr float RS_DIS_RESOLUTION = 0.005;"

mtlazaro, 2021-04-06:
So, we cannot use the rslidar_sdk with our RS16 or are we missing something?

mjseppa, 2021-04-07:
https://github.com/RoboSense-LiDAR/ros_rslidar/blob/master/rslidar_pointcloud/src/rawdata.cc#L353
https://github.com/RoboSense-LiDAR/ros_rslidar/blob/master/rslidar_pointcloud/src/rawdata.cc#L925

As RS to port the detection from old driver or just for now modify the constant for 1cm, assuming you have 1cm lidar. If you measure the actual distances and readings are half the reality, you most likely have older 1cm resolution one.

HaoQChen, 2021-04-08:
hi, mjseppa is right about that. you can change setting in rs_driver->driver->decoder->decoder_base.hpp line 62

https://github.com/RoboSense-LiDAR/rs_driver/blob/651fdd8ce94acde7cad97068b81a60f6bac895af/src/rs_driver/driver/decoder/decoder_base.hpp#L62

to 

`constexpr float RS_DIS_RESOLUTION = 0.01;`

to see if it become right

mtlazaro, 2021-04-13:
Hi all,
by changing that const variable to 0.01 we are able to use the old RS16 with the new driver returning proper distances.

closed 2021-04-13 by mtlazaro