RoboSense-LiDAR/rslidar_sdk #28: Protobuf Fails

pcarrascof, 2021-04-12:
Hello,

I am writing to you since I am using your driver with one of your sensors (RS32). I am using the Ubuntu 18 operating system with Ros Melodic. I have configured the driver so that my personal computer is the one that sends the information and an instance of Amazon Web Service is the one that consumes this information. They are connected through a VPN, with AWS being the master in the entire operation.
The driver configuration has been set as indicated in the document "how_to_use_protobuf_function.md" provided with the sdk.

The problem I find is that no sending or receiving of data is established.

Sender:
************************************************** ******
********** **********
********** RSLidar_SDK Version: v1.3.0 **********
********** **********
************************************************** ******
process [rviz-2]: started with pid [15697]
-------------------------------------------------- ----
Receive Packets From: Online LiDAR
Msop Port: 6699
Difop Port: 7788
-------------------------------------------------- ----
-------------------------------------------------- ----
Send Packets To: Protobuf-UDP
Msop Port: 60022
Difop Port: 60023
Target IP: X.X.X.X
-------------------------------------------------- ----
RoboSense-LiDAR-Driver is running .....

Receiver:
************************************************** ******
********** **********
********** RSLidar_SDK Version: v1.3.0 **********
********** **********
************************************************** ******
-------------------------------------------------- ----
Receive Packets From: Online LiDAR
Msop Port: 6699
Difop Port: 7788
-------------------------------------------------- ----
-------------------------------------------------- ----
Send PointCloud To: ROS
PointCloud Topic: / rslidar_points
-------------------------------------------------- ----
RoboSense-LiDAR-Driver is running .....
ERRCODE_MSOPTIMEOUT
ERRCODE_DIFOPTIMEOUT
ERRCODE_MSOPTIMEOUT
ERRCODE_MSOPTIMEOUT
ERRCODE_DIFOPTIMEOUT
ERRCODE_MSOPTIMEOUT
ERRCODE_MSOPTIMEOUT
ERRCODE_DIFOPTIMEOUT
ERRCODE_MSOPTIMEOUT
ERRCODE_MSOPTIMEOUT
ERRCODE_DIFOPTIMEOUT
ERRCODE_MSOPTIMEOUT
ERRCODE_MSOPTIMEOUT
ERRCODE_DIFOPTIMEOUT
ERRCODE_MSOPTIMEOUT
ERRCODE_MSOPTIMEOUT

I have checked the correct sensor operation using ROS and I have been able to observe the point cloud without any problem.

HaoQChen, 2021-04-20:
rslidar_sdk just send protobuf message to another place, it will not forward lidar's UDP to `receiver`. 

So what you need to do in `Receiver` is to capture protobuf message and decode them to normal message like `pcl::PointCloud<pcl::PointXYZI>`

You could search some tutorials about protobuf, rslidar_sdk‘s protobuf message are defined in https://github.com/RoboSense-LiDAR/rslidar_sdk/tree/release/src/msg/proto_msg

HaoQChen, 2021-04-20:
could you share your `config.yaml` ?

it seem that, your receiver did not set `msg_source` to receive message from protobuf

adamanov, 2022-02-23:
You might need to change your internet connection settings, and after that you can properly connect to the lidar

[image]

arwinder-jaspal, 2023-03-02:
> You might need to change your internet connection settings, and after that you can properly connect to the lidar
> 
> [image]

I have the exact same setting and the lidar is not working

CQ-Quest, 2023-05-10:
same problem encounter. If you have confirmed that all the configurations are correct and there is no problem with the lidar itself, try resetting it by turning off and on the power. That's how I solved the problem.

lonlonago, 2023-09-06:
port num is error, reset it in config file

closed 2026-04-10 by AmonYang2471