RoboSense-LiDAR/rslidar_sdk #33: Release as package into ROS

bochen87, 2021-04-22:
Hi, are there plans to release as a packge into ROS?

HaoQChen, 2021-04-23:
What's the benifit to do that?

easy to install?

bochen87, 2021-04-28:
yes, basically you can then install as a package instead of compilation from source.

HaoQChen, 2021-05-12:
Compilation from source will give user more option to choose what they need. For example, some users dont need ROS and they dont even have the environment of ROS while some need that to communicate.

In our test, the compilation of rslidar_sdk just costs a few seconds, is that unacceptable?

Or it is hard to clone the code from github? The transmission speed from github is slow?

Or you need a deb to install to other computers? You can download source code and compile it to a deb,   `bloom-generate`, `fakeroot` will help you with that. In this way, you can totally offline.

Timple, 2021-06-10:
Yes, ease of installation is definitely important!

> For example, some users dont need ROS

They can still install from source of course.

> Or it is hard to clone the code from github? 

Yes, you need to clone. And modify `CMakeLists.txt` file. And move a `package.xml` file. 

So all our robots that would use this lidar would need to do this. But also in CI: instead of `rosdep install` you need to perform the steps above. Which are a-typical. And they would need to be programmed into every repository using this dependency.

_Alternative: we could generate the .deb ourselves. But then we should host it somewhere, download it from a custom location on every CI run etc._

While on the other hand, a bloom release is a process which would take a [couple of hours](http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty) to configure. And as of then, everybody using ROS would have the benefit of avoiding the above :slightly_smiling_face: 
 
As we have an interest in this lidar, I would be happy to help in this process.

HaoQChen, 2021-06-11:
> Yes, ease of installation is definitely important!
> 
> > For example, some users dont need ROS
> 
> They can still install from source of course.
> 
> > Or it is hard to clone the code from github?
> 
> Yes, you need to clone. And modify `CMakeLists.txt` file. And move a `package.xml` file.
> 
> So all our robots that would use this lidar would need to do this. But also in CI: instead of `rosdep install` you need to perform the steps above. Which are a-typical. And they would need to be programmed into every repository using this dependency.
> 
> _Alternative: we could generate the .deb ourselves. But then we should host it somewhere, download it from a custom location on every CI run etc._
> 
> While on the other hand, a bloom release is a process which would take a [couple of hours](http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty) to configure. And as of then, everybody using ROS would have the benefit of avoiding the above 
> 
> As we have an interest in this lidar, I would be happy to help in this process.

copy that, we will seriously consider this matter

Timple, 2021-07-23:
#48 Is all that is required for this I guess.

Now only the tutorial needs to be followed: http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease

Timple, 2021-11-29:
Silence means consent?

Timple, 2022-01-03:
Via email I got the approval to release this package. I will hold for a week to see if #48 and #52 will be merged first.