RoboSense-LiDAR/rslidar_sdk #36: how to convert RS_Ruby cloud to range image?

YangSiri, 2021-05-12:
Hi, 

Could you please tell me how to convert 128 RS_Ruby point cloud into range image? Is there any ring information or should i calculate it through vertical angle? What's the actual horizontal angle resolution?

Looking forward to your help ;)

HaoQChen, 2021-05-13:
umm, we haven't tried this so far

horizontal angle resolution is 0.2 degree

you could find some information about pinhole model of camera

YangSiri, 2021-05-13:
Thanks, @HaoQChen

closed 2021-05-13 by YangSiri

FishInWave, 2021-11-15:
> 谢谢，@HaoQchen
> 
> 当我使用预定义的点云格式**RsPointXYZIRT**并 通过**pcl::fromROSMsg(*laserCloudRsPointXYZIRT)**将其从 ros msg 转换为点云 时的另一个问题**；**，出现以下错误。你能告诉我哪里可能出错吗？
> 
> [image]

Have you solved this problem?