RoboSense-LiDAR/rslidar_sdk #38: ROS1 packets produced by the node have wrong information and recorded bags have wrong metadata.

mjseppa, 2021-05-31:
The packet topics produced by the driver do not contain all the information needed for proper operation.
When bags are recorded, the description and md5sum of the packet messages are d41d8cd98f00b204e9800998ecf8427e (empty string) and some programs do not operate on the bags.
rosbag check on a bagfile gives following:

```

WARNING: For type [rslidar_msgs/rslidarScan] stored md5sum [ff6baa58985b528481871cbaf1bb342f] has invalid message definition."                                                            
WARNING: For type [rslidar_msgs/rslidarPacket] stored md5sum [1e4288e00b9222ea477b73350bf24f51] has invalid message definition."                                                          
The following migrations need to occur:                                                      
 * From: rslidar_msgs/rslidarScan [d41d8cd98f00b204e9800998ecf8427e]                         
   To:   rslidar_msgs/rslidarScan [ff6baa58985b528481871cbaf1bb342f]                         
    1 rules missing:                                                                         
     * From: rslidar_msgs/rslidarScan [d41d8cd98f00b204e9800998ecf8427e]
       To:   rslidar_msgs/rslidarScan [ff6baa58985b528481871cbaf1bb342f]
 * From: rslidar_msgs/rslidarPacket [d41d8cd98f00b204e9800998ecf8427e]
   To:   rslidar_msgs/rslidarPacket [1e4288e00b9222ea477b73350bf24f51]
    1 rules missing:
     * From: rslidar_msgs/rslidarPacket [d41d8cd98f00b204e9800998ecf8427e]
       To:   rslidar_msgs/rslidarPacket [1e4288e00b9222ea477b73350bf24f51]

```

The rules can be generated to fix the bag and are attached, but should be fixed on the node to avoid this.

[fix_packets-bmr.txt](https://github.com/RoboSense-LiDAR/rslidar_sdk/files/6568502/fix_packets-bmr.txt)

rosbag play for example will work with the node listening on packets, but for example bag filtering cannot be done as the message definition is broken.

HaoQChen, 2021-06-01:
Did you install the old driver `ros_rslidar` in your computer?

mjseppa, 2021-06-01:
Only the _msgs ros1 part so the message can be checked out.

mjseppa, 2021-06-01:
Also comparing the rslidar_sdk  msg/ros_msg/lidar_packet_ros.h and one generated from the old rslidar_msgs seem to be almost identical, md5sums match in the headers.

mtlazaro, 2021-09-07:
Hi,
I'm having a similar problem.
I'm using the last rslidar_sdk and never installed the previous ros_rslidar driver. 

I'm having the following errors when trying to use rosbag tools (for example rosbag check) :
```
WARNING: For type [rslidar_msgs/rslidarPacket] stored md5sum [1e4288e00b9222ea477b73350bf24f51] has invalid message definition."
WARNING: For type [rslidar_msgs/rslidarScan] stored md5sum [ff6baa58985b528481871cbaf1bb342f] has invalid message definition."
The following migrations need to occur:
 * From: rslidar_msgs/rslidarPacket [d41d8cd98f00b204e9800998ecf8427e]
   To:   Unknown
    1 rules missing:
     * From: rslidar_msgs/rslidarPacket [d41d8cd98f00b204e9800998ecf8427e]
       To:   Unknown
 * From: rslidar_msgs/rslidarScan [d41d8cd98f00b204e9800998ecf8427e]
   To:   Unknown
    1 rules missing:
     * From: rslidar_msgs/rslidarScan [d41d8cd98f00b204e9800998ecf8427e]
       To:   Unknown

To generate rules, please run with -g <rulefile>
```

mjseppa, 2021-09-08:
@mtlazaro You need to install rslidar_msgs and source them, then you will see the actual message in the "To: " field. Then you can use the rosbag fix with the file attached in earlier message (or generate your own).

You can still use the rslidar_sdk to decode the packets to pointclouds by changing the yaml file to use ROS topics instead of live lidar and use "rosbag play" as it will not care about the md5sum mismatch, but filtering is out unless you fix the md5sums.

mtlazaro, 2021-09-08:
Thanks @mjseppa, installing `ros-melodic-rslidar-msgs` have solved the "Unknown" message in the "To: " field. 
Then, using your bmr file I have been able to fix the bag.

And yes, as you mention, I was able before to decode packets with the sdk but I needed to use a bag tool to change the timestamp of another topic in the same bag and I was not able to do it because of the md5sum mismatch.