RoboSense-LiDAR/rslidar_sdk #40: Problem about flat point cloud

Richardson-Chong, 2021-06-09:
I foundd closed issues #20 and got that the reason why the point cloud became flat in my CP using rslidar_sdk might was that my RS16 was so old. After I added the CSV data, it became vertical but sparser than old one using old driver. I found that you gave the user in #20 a specified data and I sincerely hope that you can give me that data. Thanks for your reading!

HaoQChen, 2021-06-10:
angle.csv will not lead to the sparser point cloud, maybe there is some bug else

you could capture a 10 seconds  `.pcap` data so I can analyze for you

Richardson-Chong, 2021-06-11:
I'm so sorry that may the appearance is little diffierent than older one. The points that further becomes nearly totally red than the older on, and in rviz may each point shows smaller in rslidar_sdk which don't influnence using it.

HaoQChen, 2021-06-15:
`point shows smaller` you can set point size in rviz,

and the color of point means intensity of object, maybe your lidar's version is very old. The old version lidar's way to calculate intensity is different from the new one

closed 2021-07-30 by baoxianzhang