RoboSense-LiDAR/rslidar_sdk #41: How to record point cloud data into a PCAP file programmatically, just like RSView

jonbakerfish, 2021-06-21:
Hi, how to record point cloud data into a PCAP file programmatically, just like RSView?
This SDK provides examples of sending package proto or decoded point cloud proto, and we can save these protos separately for each scan. However, this will create lots of files.

HaoQChen, 2021-06-25:
1. start a process to record, you can use `tcpdump` in linux
2. use `libpcap`, this may require some time to understand this lib

RS-Alex, 2021-12-07:
> Hi, how to record point cloud data into a PCAP file programmatically, just like RSView? This SDK provides examples of sending package proto or decoded point cloud proto, and we can save these protos separately for each scan. However, this will create lots of files.

The easiest way is to install a Wireshark tool on your computer then it will start capturing UDP packets, and you can save the UDP data as a pcap file.

closed 2021-12-07 by RS-Alex