RoboSense-LiDAR/rslidar_sdk #44: [ROS1]timestamp of each packet of /rslidar_packets is zero

BenjaminChi007, 2021-07-01:
Hi,

We are trying to verify new rslidar SDK from RoboSense.
The version of the SDK is the latest release version which is tagged as v1.3.0.
We figured out that timestamp of each packet of /rslidar_packets is zero, which blocked our further verification.
Could you kind give a hand on it? Thanks in advance.

```
# rostopic echo -n 1 /rslidar_packets | more
header: 
  seq: 564
  stamp: 
    secs: 1483229108
    nsecs: 801424026
  frame_id: "/rslidar"
packets: 
  - 
    stamp: 
      secs: 0
      nsecs:         0
    data: [85, 170, 5, 90, 0, 1, 0, 0, 249, 252, 255, 2, 0, 6, 149, 160, 0, 1, 0, 0, 0, 0, 88, 104, 71, 180, 0, 10, 182, 136, 6, 0, 0, 0, 0, 0, 0, 0, 19, 16, 0, 0, 255, 238, 1, 100, 
```
BTW, the legacy https://github.com/RoboSense-LiDAR/ros_rslidar works well on the issue.

Looking forward to your reply. Thanks.

Regards,
Ben

BenjaminChi007, 2021-07-02:
Never mind，we have fixed the issue by ourselves.

closed 2021-07-02 by BenjaminChi007

tbroed, 2021-11-24:
I am having the same issue. How did you fix it?

BenjaminChi007, 2021-11-24:
> I am having the same issue. How did you fix it?
The default ros_msg in the repo may not match ROS what u are using now.
Try to rebuild the following header on ur ROS platform and replace it.
https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/release/src/msg/ros_msg/lidar_packet_ros.h
Hope it will be helpful.

tbroed, 2021-11-24:
How do I rebuild the header?
In the file you linked, it said that it is "generated by gencpp from file rslidar_msgs/rslidarPacket.msg".
Since rslidar_msgs is not part of this repo, do I have to use https://github.com/RoboSense-LiDAR/ros_rslidar ? (and how would I transfer it into this repo than)