RoboSense-LiDAR/rslidar_sdk #46: Multilidar Time Synchronization

sif95, 2021-07-06:
I am trying to connect multiple lidars to synchronize with the same GPS time. Is it possible to synchronize multiple lidar with the same timestamp?

tbroed, 2021-07-12:
Did you find an answer to this question?

mjseppa, 2021-10-05:
We are running PPS to couple RS-16s, then use use_lidar_clock. Atleast RS-16 has FW bug where it triggers on falling edge so we had to add 1.1s time offset with 100ms PPS pulse.

RS-Alex, 2021-12-07:
> I am trying to connect multiple lidars to synchronize with the same GPS time. Is it possible to synchronize multiple lidar with the same timestamp?

Robosense can offer a sync board that allows multiple lidars to run with the same GPS signal. Please contact your sales manager for more information.

closed 2021-12-07 by RS-Alex