RoboSense-LiDAR/rslidar_sdk #50: Failed to find match for field 'intensity'.

wwtinwhu, 2021-07-16:
Hello, I use pcl_conversion to convert sensor_ msgs::PointCloud2 type to pcl::PointCloud<pcl::PointXYZI> type. However, in the console , it output " failed to find match for fileld 'intensity' ".  As a result, intensity value is 0 in PointXYZI. I check the /rslidar_points in Rviz.  It can visualize in intensity  mode.  Anybody know to fix this? Thank you.

FishInWave, 2021-09-10:
The rs-driver uses uint8 to store the intensity data, while ROS and PCL using float.

wienans, 2022-08-10:
Hi I am running into similar issues. As PCL Velodyne and other LiDAR manufactures seem to publish intensity in float. Wouldn’t it make sense to use it instead of the unit8_t? 
then all the conversion functions of PCL would easily work and one wouldn’t need to iterate over all points to copy the point cloud into a other format. 
And instead of defining float intensity it might be even better to use the PCL macro  PCL_ADD_INTENSITY like you use the PCL_ADD_POINT4D.

closed 2022-08-16 by wwtinwhu