RoboSense-LiDAR/rslidar_sdk #53: Compile with ROS catkin tools

ManChrys, 2021-07-26:
Hello,

in the instractions on step 2 said rename package_ros1.xml to package.xml but when im catkin_make i have the following errors:

-- ROS Found, Ros Support is turned On!
=============================================================
CMake Error: File /media/geo/d057a6b6-f590-4870-a4f8-13b392d0f9eb/geo/catkin_ws/src/rslidar_sdk/package_ros1.xml does not exist.
CMake Error at rslidar_sdk/CMakeLists.txt:34 (configure_file):
  configure_file Problem configuring file

baoxianzhang, 2021-07-30:
configure_file need package_ros1.xml, so for now you don't need to copy and rename the file, the configure_file will do it for you.  
But configure_file does't work when use catkin_make directly, we consider to remove the configure_file function.

ManChrys, 2021-07-30:
@baoxianzhang thank you my friend.

closed 2021-07-30 by ManChrys