RoboSense-LiDAR/rslidar_sdk #59: How to define IP and MSOP DIFOP?

foreignrobot, 2021-09-07:
Hello,
I have the same issue as this: https://github.com/RoboSense-LiDAR/rslidar_sdk/issues/17#issue-779556399

I'm trying to connect multiple M1 to my system. RSView is mentioned in the documentation but the software does not seem to support the M1.

The sensors will be used on a system with Ubuntu 20.04 with ROS 1 but I also have access to a Windows machine if necessary for the configuration steps.

Thanks!

RS-Alex, 2021-12-03:
M1 LiDARs need a software tool called mems_flash.exe to define IP and the ports. Please contact alex.guo@robosense.cn for this tool.

closed 2021-12-07 by RS-Alex

TimRehb, 2023-11-13:
The email address seems outdated. Where can I find a recent contact?
Or is the software publicly available?

I bought several M1 sensors and need to confiure those properly.

Thanks in Advance!

richibrics, 2025-03-20:
Hi, did you manage to change the LiDAR IP address ?

TimRehb, 2025-03-24:
Hey, yes I did manage to change LiDAR IP adress as well as MSOP/DIFOP; it works by using the Robosense "little-robo" tool. Sometimes it can be a bit of a struggle to get the connection though...

hadisoussi1, 2025-03-31:
Hello @TimRehb, Could you please share how to download this tool? I'm having trouble finding it online.
Thank you!

TimRehb, 2025-03-31:
I got this tool via a Wetransfer download link by the support of Robosense. Unfortunately, the download link is not valid anymore. I´d suggest, you email robosense and ask for a download link of the "Little Robo M1 Customer" tool.

https://store.robosense.ai/pages/contact