RoboSense-LiDAR/rslidar_sdk #67: Compiling directly

guragamb, 2021-12-07:
Hi, I'm trying to setup my RS-Helios and trying to setup the SDK  and get the following errors when compiling directly (4.1 from README): 
```
user@ubuntu:~/rslidar_sdk/build$ cmake .. && make -j4
=============================================================
-- ROS Found, Ros Support is turned On!
=============================================================
=============================================================
-- ROS2 Not Found, Ros2 Support is turned Off!
=============================================================
=============================================================
-- Protobuf Found, Protobuf Support is turned On!
=============================================================
-- COMPILING /home/user/rslidar_sdk/src/msg/proto_msg/packet.proto USING /usr/bin/protoc
-- COMPILING /home/user/rslidar_sdk/src/msg/proto_msg/point_cloud.proto USING /usr/bin/protoc
-- COMPILING /home/user/rslidar_sdk/src/msg/proto_msg/scan.proto USING /usr/bin/protoc
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

=============================================================
-- Cmake run for UNIX GNU Compiler
=============================================================
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   date_time
--   regex
=============================================================
-- rs_driver Version : v1.3.0
=============================================================
=============================================================
-- POINT_TYPE is XYZI
=============================================================
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/rslidar_sdk/build
Scanning dependencies of target rslidar_sdk_node
[ 16%] Building CXX object CMakeFiles/rslidar_sdk_node.dir/src/msg/proto_msg/scan.pb.cc.o
[ 33%] Building CXX object CMakeFiles/rslidar_sdk_node.dir/node/rslidar_sdk_node.cpp.o
[ 50%] Building CXX object CMakeFiles/rslidar_sdk_node.dir/src/msg/proto_msg/packet.pb.cc.o
[ 66%] Building CXX object CMakeFiles/rslidar_sdk_node.dir/src/manager/adapter_manager.cpp.o
[ 83%] Building CXX object CMakeFiles/rslidar_sdk_node.dir/src/msg/proto_msg/point_cloud.pb.cc.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
CMakeFiles/rslidar_sdk_node.dir/build.make:86: recipe for target 'CMakeFiles/rslidar_sdk_node.dir/src/manager/adapter_manager.cpp.o' failed
make[2]: *** [CMakeFiles/rslidar_sdk_node.dir/src/manager/adapter_manager.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:547: recipe for target 'CMakeFiles/rslidar_sdk_node.dir/all' failed
make[1]: *** [CMakeFiles/rslidar_sdk_node.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
``` 
Thanks for taking a look at this!

RS-Alex, 2021-12-07:
Did you try compiling with ROS catkin tools?

guragamb, 2021-12-27:
@RS-Alex Hi just getting back to this...I did not try it with ROS catkin yet. Should I just make a folder for catkin_ws and in there have build, devel, src and put the rslidar_sdk project in the src? Is no build file? Thanks!

guragamb, 2021-12-28:
Just did this `sudo ln -s /usr/lib/python2.7/dist-packages/vtk/libvtkRenderingPythonTkWidgets.x86_64-linux-gnu.so /usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so` and it resolved the VTK issue that was giving the make error. It works now, thanks!

closed 2021-12-28 by guragamb