RoboSense-LiDAR/rslidar_sdk #68: RS32 point cloud looks flat

NozomiWyane, 2021-12-07:
[image]
this is how it looks like when I run sdk in ubuntu ros.
[image]
The config is list below.
[image]
Please help me out.

RS-Alex, 2021-12-08:
[angle.csv](https://github.com/RoboSense-LiDAR/rslidar_sdk/files/7673947/angle.csv)
Please add the angle path parameter in your config file. You can find the detailed information about "angle path" parameter in driver "hidden_parameter_intro" file.
The angle.csv file is attached

NozomiWyane, 2021-12-08:
> 

Thanks for your support. I try this csv in my path then the point cloud change to this below.
[image]

My config.yaml is list below:
`common:
  msg_source: 1                                         #0: not use Lidar
                                                        #1: packet message comes from online Lidar
                                                        #2: packet message comes from ROS or ROS2
                                                        #3: packet message comes from Pcap file
                                                        #4: packet message comes from Protobuf-UDP
                                                        #5: point cloud comes from Protobuf-UDP
  send_packet_ros: false                                #true: Send packets through ROS or ROS2(Used to record packet)
  send_point_cloud_ros: true                            #true: Send point cloud through ROS or ROS2
  send_packet_proto: false                              #true: Send packets through Protobuf-UDP
  send_point_cloud_proto: false                         #true: Send point cloud through Protobuf-UDP
  pcap_path: /home/robosense/lidar.pcap                 #The path of pcap file

lidar:
  - driver:
      lidar_type: RS32            #LiDAR type - RS16, RS32, RSBP, RS128, RS80, RSM1, RSHELIOS
      frame_id: /rslidar           #Frame id of message
      msop_port: 6699              #Msop port of lidar
      difop_port: 7788             #Difop port of lidar
      start_angle: 0               #Start angle of point cloud
      end_angle: 360               #End angle of point cloud 
      min_distance: 0.2            #Minimum distance of point cloud
      max_distance: 200            #Maximum distance of point cloud
      use_lidar_clock: false       #True--Use the lidar clock as the message timestamp
                                   #False-- Use the system clock as the timestamp  
      angle_path: /home/cdp/ws_lidar/src/rslidar_sdk/config/angle.csv
      split_frame_mode: 1
      cut_angle: 0
      num_pkts_split: 1
      wait_for_difop: true
      saved_by_rows: false
      multi_cast_address: 0.0.0.0
      x: 0
      y: 0
      z: 0
      roll: 0
      pitch: 0
      yaw: 0
    ros:
      ros_recv_packet_topic: /rslidar_packets          #Topic used to receive lidar packets from ROS
      ros_send_packet_topic: /rslidar_packets          #Topic used to send lidar packets through ROS
      ros_send_point_cloud_topic: /rslidar_points      #Topic used to send point cloud through ROS
    proto:
      point_cloud_recv_port: 60021                     #Port number used for receiving point cloud 
      point_cloud_send_port: 60021                     #Port number which the point cloud will be send to
      msop_recv_port: 60022                            #Port number used for receiving lidar msop packets
      msop_send_port: 60022                            #Port number which the msop packets will be send to 
      difop_recv_port: 60023                           #Port number used for receiving lidar difop packets
      difop_send_port: 60023                           #Port number which the difop packets will be send to 
      point_cloud_send_ip: 127.0.0.1                   #Ip address which the point cloud will be send to 
      packet_send_ip: 127.0.0.1                        #Ip address which the lidar packets will be send to
`
I think it still doesn't work normally. Please help?

RS-Alex, 2021-12-08:
Can I ask if you were playing with the RS-Helios (New 32 beam) or RS-32 (old 32 beam) unit?
Can I please have your SN number?

NozomiWyane, 2021-12-08:
Im using the old 32 beam and SN is161181401417

RS-Alex, 2021-12-08:
It seems like this lidar is faulty, please try run this lidar in the RS-View software and if the performance is still the same, please contact our sales manager for fix/replacement issue.

NozomiWyane, 2021-12-08:
There is no points in the RSview. But I can still see points in ros sdk. Is there something wrong in my config of RSview?
[image]
Three csv files are also in this dir

RS-Alex, 2021-12-08:
Please select RSlidar32. You also need to make sure your port is set correctly. (Tools -> data port setting)
[image]

NozomiWyane, 2021-12-08:
[image]
Now points cloud looks like this.
I found a fault diagnosis and result like this
[image]

RS-Alex, 2021-12-08:
Please contact our sales manager for fix/replacement issues.

closed 2021-12-29 by RS-Alex