RoboSense-LiDAR/rslidar_sdk #69: Vertical Angles from Difop not parsed correctly - Flat Pointcloud - RS128 (RS-Ruby)

renezurbruegg, 2021-12-07:
I have a similar problem as many other people (#68 #20 #40 #24), meaning the ROS pointcloud is flat.

I checked and the difop package arrives successfully and the calibration callback is executed [Code Line In Driver](https://github.com/RoboSense-LiDAR/rs_driver/blob/651fdd8ce94acde7cad97068b81a60f6bac895af/src/rs_driver/driver/decoder/decoder_RS128.hpp#L228).
However the angle is never calibrated, as this function always terminates [Code Line In Driver](https://github.com/RoboSense-LiDAR/rs_driver/blob/e9d8d761455b53b8276a10dc4831501e09557c74/src/rs_driver/driver/decoder/decoder_base.hpp#L663)

[image]

Any help would be appreciated!

RS-Alex, 2021-12-08:
[angle.csv](https://github.com/RoboSense-LiDAR/rslidar_sdk/files/7673965/angle.csv)
Please try the same method with #68 , the angle.csv for RS128 is attached.

renezurbruegg, 2021-12-08:
Great, this fixed the issue.
Thanks a lot for the fast response!

closed 2021-12-08 by renezurbruegg