RoboSense-LiDAR/rslidar_sdk #71: rs view from lidar-m1 download page not showing point cloud, also same with ubuntu version of rs-view.

Rinntrah, 2021-12-27:
Hi, struggling with this problem for so long now... Rs view from lidar-m1 download page is not showing point cloud, despite correct msop/difop port configuration (6699,7788). Packets are received by 192.168.1.200 computer(from wireshark). ALSO Same version of rsview for ubuntu does not show any pointcloud nor temperature nor the mac adress of lidar. Can't find any help because robosense ignores all of my emails.

Is there like a secret guide to configuring lidar m1? Why official software is not working despite following all steps in official instruction?
Packets are sent from lidar to 192.168.1.200 computer, but rs-view software struggles to recognize the lidar...
**Has anyone met with problem like above mentioned?**

RS-Alex, 2021-12-28:
Hi, sorry for the issue you have, it is seen from your comment that the most likely reason is the firewall is on, since the wireshark captured the packets correctly the lidar itself is working ok. Please try trun off the fire wall, and if it's still not working, please send the pcap file to alex.guo@robosense.cn and I will help with further diagnose.

closed 2021-12-29 by RS-Alex

RS-Alex, 2022-02-23:
Hi Michal 
Please contact alex.guo@robosense.cn, I will send you the B2 driver.

deepuav, 2023-11-12:
@RS-Alex  
 I will need the B2 driver too

deepuav, 2023-11-12:
weichat 1317502829

Rinntrah, 2024-05-28:
Hey, the link has expired, and we need again at our UITM university rsview,
for experiments, can you send me a new one rsview for my B2 unit? Also can
you send Ubuntu rsview for that B2 unit?
Michał
Furgał