RoboSense-LiDAR/rslidar_sdk #74: Multiple LiDAR launch from single workspace.

ajay1606, 2022-01-21:
Hello ,

We are currently using 3 Lidars (Ruby, 2 BPearl), is there any way that I can modify the ROS rslidar_sdk_node to launch 3 Lidars from the same computer.

APpreciate your response.

Regards,
Ajay

RS-Alex, 2022-01-24:
Yes you can use multiple lidars on the same computer.
You can find the instructions in SDK. Path below:
\src\rslidar_sdk\doc\howto\how_to_use_multi_lidars.md

ajay1606, 2022-01-24:
@RS-Alex Thanks Alex for your quick response. But there is an issue with configuring BPearl Port Number to launch multiple LiDARS. As I understood that, BPearl Models doesn't support for the web interface, I have tried with the RSView downloaded from the following link; [link](https://www.robosense.ai/en/resources-20)
After the following instructions am getting the following warning notification and am unable to set a unique port number for each lidar. I have tried with 4 units of BPearl LiDARS (Purchased recently). All 4 units have the same problem.

We greatly appreciate your kind assistance as we are looking to fix this as soon as possible. Ref screenshot attached.

Sensor Configuration Tab:
[image]

After clicking the **Get** button (There were no sensor information loading), Also Tried with selecting BP option from Drop menu.
[image]

Again after trying with **set LiDAR**
[image]

There was no configuration saved. Greatly appreciate your response.

Regards,
Ajay

RS-Alex, 2022-01-24:
Hi Ajay
Before parameter settings please check the version of your BPearl:
[image]
You can only set lidar by RSView if the revision is 2.1.0. If the revision is 3.0, you need to go to browser webpage and put in lidar IP to change all the parameters.
If the revision version is ok, and you are still having problem connecting lidars, please contact alex.guo@robosense.cn, we can make online meeting for further diagnose.

ajay1606, 2022-01-24:
@RS-Alex Thanks for your quick response. All 4 units were V2.1.0 versions. And figured out the reason for that cause. 
After verifying the data in Wireshark, it is understood that there is a mismatch with **difop_port** and **msop_port** numbers. After fixing that correctly as it mentioned in Wireshark, Get and SetLiDAR options working well.

Many thanks for your quick responses and continued support.  

Regards,
Ajay

RS-Alex, 2022-01-24:
No problem, please contact if you have any more questions:)

closed 2022-01-24 by RS-Alex