RoboSense-LiDAR/rslidar_sdk #75: 1.3.1 release

Timple, 2022-01-27:
I noticed the new 1.3.1 release.

However the release branch is not there yet. Also the `package.xml`s in the 1.3.1 release still point to 1.3.0.

Could these issues be fixed?

_Tagging @ronzheng1006 who did the release it seems?_

RS-Alex, 2022-02-09:
@ronzheng1006 Hi Ron can you please help to fix the issue?

Timple, 2022-02-17:
Any update here? Seems like a trivial effort. Having the config file as a parameter would be nice.
But our tooling breaks on the tag mismatch with the package.xml

RS-Alex, 2022-02-23:
Hi @Timple Ron is working on this and will update shortly. Sorry for the delay.

Timple, 2022-10-11:
So now the release branch is on `1.3.2` but release `1.5.7` is already made.

Isn't it easier to drop the release branch or drop the releases? Now they keep getting out-of-sync.