RoboSense-LiDAR/rslidar_sdk #76: ROS2 implemenation really slow

Timple, 2022-01-31:
Using the `rslidar_sdk` package for ROS1 gives fine results.

However in ROS2 the update rate is really show (6hz). Using two Bpearls.
This leads to timestamps going quickly into the past.

Any else experiencing this issue?

ajay1606, 2022-02-08:
@Timple , Agree with you, output rate showing almost 6 hz with ROS2. 
In my case, am working with ;

Ubuntu 20.04 (Galactic)
Sensor : BPearl

[image]

RVIZ Visualization looks like publish rate is fine while running only sensor terminal, but when start other terminal with `ros2 bag reocrd` or `ros2 topic hz ` it appears to slow down ! 

Appreciate any suggestion to fix this.

Regards,
Ajay

RS-Alex, 2022-02-09:
Hello @Timple and @ajay1606 , thanks for the feedback. We have reported this issue and will look into the solution.

ajay1606, 2022-02-17:
@RS-Alex  Any possible fix available?

RS-Alex, 2022-02-17:
> @RS-Alex Any possible fix available?

The first stage conclusion is related to lack of the computer calculation ability. In ROS2 there is higher reqirement to the computer than ROS1. Please try modifying the DDS bandwidth and check if it can get better.

Timple, 2022-02-17:
My computer wasn't breaking a sweat (according to `htop`). So pure CPU wise I was still good.

> Please try modifying the DDS bandwidth and check if it can get better.

Do you have a suggestion on how to do this? Can't find bandwidth related settings in de guide: https://docs.ros.org/en/galactic/How-To-Guides/DDS-tuning.html

@RS-Alex where you able to reproduce the issue?

ajay1606, 2022-02-17:
@RS-Alex, Thanks for your response. And i agree with @Timple . In my case i tested LiDAR with quite high performing computer with halting all other processes. Also frequency variation observed even with the BPearl model lidars which is not quite dense enough to cause bandwidth issues.

And, FYI I have tried with 64 channel lidar (Hesai), it's working well at 10hz. (Platform same)

Regards,
Ajay

hengjiUSTC, 2022-03-06:
Same result observed in Ros2 galactic. We have around 10 modules running in the same time, after we start bag record, rslidar frequency drops to 7hz. Velodyne is stable in the same test environment.
We dig in at: https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/release/src/adapter/point_cloud_ros_adapter.hpp#L114
The time tag of LidarPointCloudMsg& msg is stable (10hz). Execution time of above line is 7hz and cause final output frequency drops. Seems thread_pool is not executing committed task immediately when system load is high.

hengjiUSTC, 2022-03-06:
> > @RS-Alex Any possible fix available?
> 
> The first stage conclusion is related to lack of the computer calculation ability. In ROS2 there is higher reqirement to the computer than ROS1. Please try modifying the DDS bandwidth and check if it can get better.

We'v tried in crease band width and even change DDS, and they all won't not help. Considering other modules and Velodyne does not drops frequency, I dont think the problem is in Ros2 side. Also please check my above comment, its more like problem in rslidar driver.

RS-Alex, 2022-03-07:
> > > @RS-Alex Any possible fix available?
> > 
> > 
> > The first stage conclusion is related to lack of the computer calculation ability. In ROS2 there is higher reqirement to the computer than ROS1. Please try modifying the DDS bandwidth and check if it can get better.
> 
> 1. We'v tried in crease band width and even change DDS, and they all won't not help. Considering other modules and Velodyne does not drops frequency, I don't think the problem is in ROS2 side. Also please check my above comment, its more like problem in rslidar driver.
> 2. Also we found another problem related to thread_pool. When system load is high, time tag ordering of output pointcloud will sometime get wrong. After we change( https://github.com/RoboSense-LiDAR/rs_driver/blob/1ed9b29388742e01e6652903838a45dd3e5d2cbf/src/rs_driver/utility/thread_pool.hpp#L39 )thread number to 1, the problem is solved. We feel the way rslidar use thread_pool is likely lead to unstable output.

@ronzheng1006 Any updates on this issue?
Please also check the thread_pool issue, thanks!

arics2020, 2022-03-22:
Same story for Robosense M1, on Jetson XAVIER, Ubuntu 18, with ROS2 Eloquent running only btwn. 4.5Hz- 8Hz. 

With Ros1 Melodic no problems, always 10Hz.

A transfer of the colcon workspace from SD card to home dir did not improve the situation.

[image]

hengjiUSTC, 2022-04-14:
After reducing system load, the frequency of rslidar_points is better(frequency not drop significantly).  
Below is topic hz result when everything is normal
```
average rate: 10.002
	min: 0.094s max: 0.106s std dev: 0.00273s window: 30
average rate: 10.001
	min: 0.095s max: 0.105s std dev: 0.00227s window: 30
average rate: 10.000
	min: 0.095s max: 0.105s std dev: 0.00229s window: 30
```

If record system with `ros2 bag record -a` (increase system load), min and max start to change and frequency drops. In some machine, the frequency drops to 7-8 hz.
```
average rate: 9.724
	min: 0.094s max: 0.122s std dev: 0.00860s window: 30
average rate: 9.741
	min: 0.093s max: 0.122s std dev: 0.00835s window: 30
average rate: 9.723
	min: 0.093s max: 0.122s std dev: 0.00876s window: 30
```

After debugging into rslidar_sdk. The result shows that publish takes longer when `ros2 bag record -a` is running.
Additional prints are added at [here](https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/main/src/adapter/point_cloud_ros_adapter.hpp#L112)
```
#point_cloud_ros_adapter.hpp
inline void PointCloudRosAdapter::sendPointCloud(const LidarPointCloudMsg& msg)
{
  RCLCPP_INFO(rclcpp::get_logger("send point cloud"), "send point cloud");
  point_cloud_pub_->publish(toRosMsg(msg));
  RCLCPP_INFO(rclcpp::get_logger("send point cloud"), "send point cloud finish");
}
```

**results**
when frequency not drop, publish takes 0.06s
```
[rslidar_sdk_node-2] [INFO] [1649932014.639356601] [send point cloud]: send point cloud
[rslidar_sdk_node-2] [INFO] [1649932014.699102137] [send point cloud]: send point cloud finish
``` 
when frequency drops, publish takes 0.1s. The observation is consistent.
```
[rslidar_sdk_node-2] [INFO] [1649931274.059146476] [send point cloud]: send point cloud
[rslidar_sdk_node-2] [INFO] [1649931274.158428471] [send point cloud]: send point cloud finish
```

I also perf the node but not sure if its helps.
when frequency not drop, rcl_publish was detect with 12.19% samples
[image]

when frequency drops, rcl_publish was detect with with 15.49% samples.
[image]

Glad to hear more feedback.

Timple, 2022-04-14:
> If record system with ros2 bag record -a (increase system load)

Not perse increase load. But in ROS2 all publishers are 'lazy'. So nothing is published when there are no subscribers.

So the rslidar node actually does more.

hengjiUSTC, 2022-04-15:
> > If record system with ros2 bag record -a (increase system load)
> 
> Not perse increase load. But in ROS2 all publishers are 'lazy'. So nothing is published when there are no subscribers.
> 
> So the rslidar node actually does more.

I probably not making clear on test setup, rslidar is tested with other modules running in the same time. So when not recording, there are still several modules using lidar result, laze publish does not happen. 
When recording using `record -a`, the lo band width increased as output of other modules are all recorded (**increase system load, publish takes longer, frequency drop)**.

bmon result:
Frequency not drop, no record, 10 modules running.
[image]

Frequency drop,  with record -a, 10 modules running.
[image]

jimmyshe, 2022-04-19:
I encounter the same issue today. I am using ros 2 foxy on intel system. When I satert the node using cyclone dds and have two nodes subscribe  for point cloud, the speed of two subscribers drop significantly.

ZhenshengLee, 2022-06-13:
@hengjiUSTC 

see this earlier discussion https://github.com/RoboSense-LiDAR/rslidar_sdk/issues/3 
It would help!

ZhenshengLee, 2022-06-13:
@Timple 

### tests and results

#### test1

I gave it tests with the following command

```sh
ros2 launch rslidar_sdk start.py
stress -c 3
# the tested pc has 6 cores of 3.2GHz of cpu
ros2 topic hz /rslidar_points
```

I tested it in ros2 galactic, ubuntu20.04, when there is no stress of system load, the frequency of rslidar_points canbe stable with 10Hz
But when I execute `stress -c 3` , that means a half of system load, the frequency drops a lot, to 3Hz.

#### test2

I tried to set thread priotity in `rslidar_sdk_node.cpp` with the following, and set the whole(including thread pool threads and node(participant) related dds threads), the frequency canbe increase, but will not get the maximum of 10Hz.

```cpp
set_this_thread_priority(98, SCHED_FIFO);

  std::shared_ptr<AdapterManager> demo_ptr = std::make_shared<AdapterManager>();
  demo_ptr->init(config);
  demo_ptr->start();
```

```cpp
int set_thread_priority(pid_t pid, size_t sched_priority, int policy)
{
  struct sched_param param;
  memset(&param, 0, sizeof(param));
  param.sched_priority = sched_priority;
  return sched_setscheduler(pid, policy, &param);
}

int set_this_thread_priority(size_t sched_priority, int policy)
{
  return set_thread_priority(getpid(), sched_priority, policy);
}
```

#### test3

I changed the `MAX_THREAD_NUM` to lower to `1` , then execute `stress -c 3`, the frequency can be improved, but cannot be the maximux of 10Hz

### comments

> 1. We'v tried in increase band width and even change DDS

I tried it with rmw_ecal_dynamic_cpp, and the issue has been solved, the topic frequency is stable with 10Hz

> Seems thread_pool is not executing committed task immediately when system load is high.

I changed the `MAX_THREAD_NUM` to lower to `1` , then execute `stress -c 3`, the frequency can be improved, but cannot be the maximux of 10Hz

> With Ros1 Melodic no problems, always 10Hz.

@arics2020 
In ROS1, cpu bounds can reduce the frequency of the topic, if you execute `stress` to fullfill the cpu loads 

> Restested with another Lidar Manufacturer same Problem. So it seems to be a Hardware dependent Problem as @RS-Alex said.

it must be something wrong with the dds layer of ros2

@hengjiUSTC 
> Velodyne and all our other modules are stable in the same test environment.

@ajay1606 
> And, FYI I have tried with 64 channel lidar (Hesai), it's working well at 10hz. (Platform same)

this is controversial about the independentence of hardware(driver).

ZhenshengLee, 2022-06-14:
### more tests

#### change the size of data topic

change parameter `start_angle` and `end_angle` cannot solve the issue, which means that the message size may not be crusial to the frequency of the topic

### more comments

#### the qos setting of subscriber

according to this article, the `ros2 topic hz` may output wrong results, so there is a need to create a `pointcloud subscriber` in cpp.
https://answers.ros.org/question/350753/ros2-topic-hz-provides-wrong-rate-for-larger-msgs/

ZhenshengLee, 2022-06-16:
Tested, the branch https://github.com/RoboSense-LiDAR/rslidar_sdk/tree/dev_opt , did resolve the hz dropping issue by reducing the size of the published clouddata.

Timple, 2022-06-16:
> did resolve the hz dropping issue by reducing the size of the published clouddata.

This sounds more like a workaround instead of a solution...

ZhenshengLee, 2022-06-16:
@hengjiUSTC 

Actually, in my test, the frequency of the publisher did not drop, with the testing method in #3 
the code canbe found in https://github.com/ros2driver/rslidar_sdk/tree/outdoor/release 

What I found that drops a lot is the received msg frequency of the pointcloud subscriber(ros2 topic hz and my custom `pcl_listener` in https://github.com/ZhenshengLee/ros2_shm_msgs/tree/master/src )

finally , in the branch https://github.com/RoboSense-LiDAR/rslidar_sdk/tree/dev_opt, things become OK.

Everyone should have a try.

Timple, 2023-02-08:
So this issue has been open for a year now.

Anyone found a solution that did not involve reducing the pointcloud size or the update rate?

wavespectrum, 2024-06-21:
Can you guys share your parameter files for both ros1 and ros2 so i can attempt to reproduce the issue with newer SDK version

karthik101200, 2025-04-30:
Hello
I am facing a similar issue on the robosense airy
The raw pointcloud frequency is around 10 Hz initially but starts jittering and frequency becomes unstable ranges from 5 -15 Hz sometimes and  this is visible on visualizer like foxglove
Along with this any node that subscribes to the pointcloud and publishes a topic has frequency drops i.e the published topic has a frequency about 5 Hz. I checked the node and the processing doesnt take a lot of time as the same code works for another lidar with 10 Hz. To reduce the pointcloud size i also changed the max distance which is now at around 4-5 meters

gfchen01, 2025-09-15:
Hi, I think it's a ros2 problem. On my end in ros1 noetic, even `stress -c 16` on my 16 core CPU doesn't cause frequency drop.

An easy fix is to changing the RMW_IMPLEMENTATION to Zenoh. I am using Ros2 jazzy. Follow instructions here:

https://docs.ros.org/en/jazzy/Installation/RMW-Implementations/Non-DDS-Implementations/Working-with-Zenoh.html

Never drops frames even with `stree -c 16` with this setup.

P.S.: I have tried fast_dds and cyclone_dds. Fast_dds is the worst. Cyclone_dds still worked with `stress -c 12`, it starts to drop message with higher CPU load.