RoboSense-LiDAR/rslidar_sdk #77: Unable to build workspace in ROS2

ajay1606, 2022-02-07:
Hello,

We are trying to compile ROS2 workspace to get the sensor data in ROS2 platform. But we are getting compilation error life below.
Please could you suggest me, where we are missing ?

We are able to get data in ROS, but in ROS2 we are unable to launch the workspace successfully.
After running following command, we are getting error as mentioned below;

`ros2 launch rslidar_sdk start.py`
                   

Confirmed following modification:
```
set(COMPILE_METHOD COLCON)

Rename the file package_ros2.xml in the rslidar_sdk to package.xml
```

**Error Screen message:**
[image]

Working Platform:
Ubuntu 20.04 (galactic)
ROS2 
LIDAR (BPearl)

Appreciate your kind response.

Regards,
Ajay

ajay1606, 2022-02-08:
Commit mentioned here, successfully fixed the issue. (fix: fix launch.py, adapt to ros2 galactic)

closed 2022-02-08 by ajay1606

TZECHIN6, 2022-10-31:
@ajay1606 hi, how do you able to colcon build it in your environmnet? Mine is same with you galactic in Ubuntu 20.04.

However, I encounter issue listed in here https://github.com/RoboSense-LiDAR/rslidar_sdk/issues/105

Have you also faced this issue? seem the error is not caused by the _sdk pacakge, instead its the _msg project.

ajay1606, 2022-11-01:
@TZECHIN6 At the time I just used this PR and it worked for me

`https://github.com/RoboSense-LiDAR/rslidar_sdk/commit/a58d63cf0beea36466a14968d3eae77192d7e96a`