RoboSense-LiDAR/rslidar_sdk #84: Update CMakeLists.txt

adamanov, 2022-03-18:
Update please CMakeLists.txt or  [Step 4.1](https://github.com/RoboSense-LiDAR/rslidar_sdk#41-compile-directly) in readme.
By default right now in repo [CMakeLists](https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/34ac46b398add4e86c81d4a19d764f24afd8054d/CMakeLists.txt#L8) `set(COMPILE_METHOD CATKIN)` is setted. To use `./rslidar_sdk_node` it is needed to complied with `set(COMPILE_METHOD ORIGINAL)` . Otherwise `./rslidar_sdk_node`  will not be generated inside `build` folder (it will be located somewhere in subfolders and lead to conflict)

Timple, 2022-03-18:
Hi @adamanov, would you mind trying the branch from PR #48?

I'm curious if that would solve your issue.

adamanov, 2022-03-18:
> Hi @adamanov, would you mind trying the branch from PR #48?
> 
> I'm curious if that would solve your issue.

Hello Timple, I don't have any problem right now,  I solved it by compiling with `set(COMPILE_METHOD ORIGINAL)` it works fine. 

However, I tried also your repo with branch `ros12` . It is still not generating ./rslidar_sdk_node 

```
cd rslidar_sdk
mkdir build && cd build
cmake .. && make -j4
./rslidar_sdk_node   # <---- is not generated 
```

see my terminal

```
[100%] Built target rslidar_sdk_node
adamanov@GL62M-7RDX:~/rslidar_sdk/build$ ls
atomic_configure  catkin_generated  CMakeCache.txt  cmake_install.cmake     CTestCustom.cmake    devel  Makefile  test_results
catkin            CATKIN_IGNORE     CMakeFiles      CTestConfiguration.ini  CTestTestfile.cmake  gtest  src
adamanov@GL62M-7RDX:~/rslidar_sdk/build$ ./rslidar_sdk_node
bash: ./rslidar_sdk_node: No such file or directory
adamanov@GL62M-7RDX:~/rslidar_sdk/build$ ls
atomic_configure  catkin_generated  CMakeCache.txt  cmake_install.cmake     CTestCustom.cmake    devel  Makefile  test_results
catkin            CATKIN_IGNORE     CMakeFiles      CTestConfiguration.ini  CTestTestfile.cmake  gtest  src
adamanov@GL62M-7RDX:~/rslidar_sdk/build$ cd devel/lib/
pkgconfig/   rslidar_sdk/ 
adamanov@GL62M-7RDX:~/rslidar_sdk/build$ cd devel/lib/rslidar_sdk/
adamanov@GL62M-7RDX:~/rslidar_sdk/build/devel/lib/rslidar_sdk$ ls
rslidar_sdk_node
adamanov@GL62M-7RDX:~/rslidar_sdk/build/devel/lib/rslidar_sdk$ ./rslidar_sdk_node 
********************************************************
**********                                    **********
**********    RSLidar_SDK Version: v1.3.2     **********
**********                                    **********
********************************************************
[ERROR] [1647606392.391811506]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...
[ INFO] [1647606397.577108725]: Connected to master at [localhost:11311]
[rospack] Error: package 'rslidar_sdk' not found
[librospack]: error while executing command
Config file format wrong! Please check the format(e.g. indentation) 
adamanov@GL62M-7RDX:~/rslidar_sdk/build/devel/lib/rslidar_sdk$ 
```

Cloudytosunny, 2022-03-29:
> > Hi @adamanov, would you mind trying the branch from PR #48?
> > I'm curious if that would solve your issue.
> 
> Hello Timple, I don't have any problem right now, I solved it by compiling with `set(COMPILE_METHOD ORIGINAL)` it works fine.
> 
> However, I tried also your repo with branch `ros12` . It is still not generating ./rslidar_sdk_node
> 
> ```
> cd rslidar_sdk
> mkdir build && cd build
> cmake .. && make -j4
> ./rslidar_sdk_node   # <---- is not generated 
> ```
> 
> see my terminal
> 
> ```
> adamanov@GL62M-7RDX:~/rslidar_sdk$ git branch -a
> * release
>   remotes/origin/HEAD -> origin/release
>   remotes/origin/config-param
>   remotes/origin/release
>   remotes/origin/revert-42-release
>   remotes/origin/ros12
>   remotes/origin/ros12-config-param
>   remotes/origin/ros12-dev
>   remotes/origin/yaml-cpp
> adamanov@GL62M-7RDX:~/rslidar_sdk$ git checkout ros12
> Branch 'ros12' set up to track remote branch 'ros12' from 'origin'.
> Switched to a new branch 'ros12'
> adamanov@GL62M-7RDX:~/rslidar_sdk$ git branch -a
>   release
> * ros12
>   remotes/origin/HEAD -> origin/release
>   remotes/origin/config-param
>   remotes/origin/release
>   remotes/origin/revert-42-release
>   remotes/origin/ros12
>   remotes/origin/ros12-config-param
>   remotes/origin/ros12-dev
>   remotes/origin/yaml-cpp
> adamanov@GL62M-7RDX:~/rslidar_sdk$ git submodule init
> Submodule 'src/rs_driver' (https://github.com/RoboSense-LiDAR/rs_driver.git) registered for path 'src/rs_driver'
> adamanov@GL62M-7RDX:~/rslidar_sdk$ git submodule update
> Cloning into '/home/adamanov/rslidar_sdk/src/rs_driver'...
> Submodule path 'src/rs_driver': checked out 'de731dc0ad254da72bad4bd70c8e648dfc098bf0'
> adamanov@GL62M-7RDX:~/rslidar_sdk$ mkdir build && cd build
> adamanov@GL62M-7RDX:~/rslidar_sdk/build$ cmake .. && make -j4
> -- The C compiler identification is GNU 7.5.0
> -- The CXX compiler identification is GNU 7.5.0
> -- Check for working C compiler: /usr/bin/cc
> -- Check for working C compiler: /usr/bin/cc -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> =============================================================
> -- ROS Found, Ros Support is turned On!
> =============================================================
> =============================================================
> -- ROS2 Not Found, Ros2 Support is turned Off!
> =============================================================
> -- Looking for pthread.h
> -- Looking for pthread.h - found
[... 137 lines cut ...]
> adamanov@GL62M-7RDX:~/rslidar_sdk/build$ ./rslidar_sdk_node
> bash: ./rslidar_sdk_node: No such file or directory
> adamanov@GL62M-7RDX:~/rslidar_sdk/build$ ls
> atomic_configure  catkin_generated  CMakeCache.txt  cmake_install.cmake     CTestCustom.cmake    devel  Makefile  test_results
> catkin            CATKIN_IGNORE     CMakeFiles      CTestConfiguration.ini  CTestTestfile.cmake  gtest  src
> adamanov@GL62M-7RDX:~/rslidar_sdk/build$ cd devel/lib/
> pkgconfig/   rslidar_sdk/ 
> adamanov@GL62M-7RDX:~/rslidar_sdk/build$ cd devel/lib/rslidar_sdk/
> adamanov@GL62M-7RDX:~/rslidar_sdk/build/devel/lib/rslidar_sdk$ ls
> rslidar_sdk_node
> adamanov@GL62M-7RDX:~/rslidar_sdk/build/devel/lib/rslidar_sdk$ ./rslidar_sdk_node 
> ********************************************************
> **********                                    **********
> **********    RSLidar_SDK Version: v1.3.2     **********
> **********                                    **********
> ********************************************************
> [ERROR] [1647606392.391811506]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...
> [ INFO] [1647606397.577108725]: Connected to master at [localhost:11311]
> [rospack] Error: package 'rslidar_sdk' not found
> [librospack]: error while executing command
> Config file format wrong! Please check the format(e.g. indentation) 
> adamanov@GL62M-7RDX:~/rslidar_sdk/build/devel/lib/rslidar_sdk$ cd ..
> ```

Can you run your ros master?

closed 2022-06-20 by adamanov