RoboSense-LiDAR/rslidar_sdk #89: ROS2 Perfomance improvement

simulacrus, 2022-05-13:
I found several places for improvement.

First, function [toRosMsg](https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/main/src/msg/ros_msg_translator.h#L152). It's better to publish a unique_ptr instead of an object, avoiding unnecessary copying when publishing a message

Second, the implementation of the driver as a component

ZhenshengLee, 2022-06-13:
> First, function [toRosMsg](https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/main/src/msg/ros_msg_translator.h#L152). It's better to publish a unique_ptr instead of an object, avoiding unnecessary copying when publishing a message

using std::move would be better?

https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/34ac46b398add4e86c81d4a19d764f24afd8054d/src/adapter/point_cloud_ros_adapter.hpp#L71

```cpp
point_cloud_pub_.publish(std::move(toRosMsg(msg))); 
```

ZhenshengLee, 2022-06-13:
> the implementation of the driver as a component

It should be hard for the current rslidar_sdk arch, because `PointCloudRosAdapter` inherits from `AdapterBase`.

boxanm, 2024-07-25:
Hi, is there any update on implementing a component version of the driver?

felixf4xu, 2025-02-27:
yes, composable node feature is needed, something like this:

```cpp
#include "rclcpp_components/register_node_macro.hpp"
RCLCPP_COMPONENTS_REGISTER_NODE(..)
```

```cmake
rclcpp_components_register_node(..
  PLUGIN ".."
  EXECUTABLE ..
)
```