RoboSense-LiDAR/rslidar_sdk #93: Support for old version of RS laser scanner.

chenpengxin, 2022-06-30:
I bought three RS-LiDAR-32 with manufacutre date of July, 2018.  They have been kept on the shelf for nearly four years, and now I found them and try to use them.

Firstly,
By using the old sdk [ros_rslidar](https://github.com/RoboSense-LiDAR/ros_rslidar), I could get point clouds with **correct geometry** _**but without ring and timestamp info.**_

Then, 
I attempted to use the new sdk [rslidar_sdk](https://github.com/RoboSense-LiDAR/rslidar_sdk), but unfortunately I got point clouds with geometry of flat circles, just like the issue [#68](https://github.com/RoboSense-LiDAR/rslidar_sdk/issues/68).

Further,
I add the param of **angle_path** to the config.yaml. It did took some effect: flat circles became 3-dimension circles. Like this:

[image]

I would think the potential reason for this is that the lidar version is too old. And like the discussion in other issues, _old version of lidars do not output device information via difop port_.  

The real problem is:
**The setting of angle.csv can only make flat point clouds to stand up. I guess I still need the interface to config the **curves.csv and ChannelNum.csv** in the new sdk to recover corner and edge geometry** 

Please help!

RS-Alex, 2022-08-08:
Hi please send a pcap file saved by Wireshark and send to alex.guo@robosense.cn so I can check the data for you. Thanks.

roger-lcc, 2022-08-12:
I have the same issue with rs-lidar-16.
Did you figure it out?
Thanks! @chenpengxin

protoCAD, 2022-08-28:
I am also planning to buy an used rs-lidar-16. Any news about this issue?

roger-lcc, 2022-08-31:
Through combining three csv files into one angle.csv(just add the curves.csv and channelnum.csv's data to angle.csv ), the new sdk can read the config from angle.csv. And also you should add the angle_path in config.yaml.