RoboSense-LiDAR/rslidar_sdk #95: RS16 LIDAR CONFIGURATIONS

sebastianhenaosanta, 2022-07-15:
Hey, right now i'm working with r16 system but i'm using a raspberry pi as system of acquisition, is there any way to set rotation speed, dual return .... without using rsView?.

RS-Alex, 2022-08-08:
Hi, we do have a tool under linux to support rs16 and rsbp to make parameter settings by command lines. Please email to alex.guo@robosense.cn and ask for the tool.

ted8201, 2022-08-11:
offer the tool directly here, would be nice.

a7g4, 2023-04-17:
Hi, I tried emailing but it bounced. Is there an updated contact or somewhere else this tool is available?

kosmonauta144, 2024-01-04:
I also tried to send email but got a notification that adress is unreachable or receiver cannot get the message. Will be grateful for help.