RoboSense-LiDAR/rslidar_sdk #99: Read the instructions, and then compile the error

YJXTTZ, 2022-08-19:
In file included from /home/x/workspace/src/rslidar_sdk/src/manager/node_manager.cpp:36:
/home/x/workspace/src/rslidar_sdk/src/source/source_packet_ros.hpp: In function ‘robosense::lidar::Packet robosense::lidar::toRsMsg(const RslidarPacket&)’:
/home/x/workspace/src/rslidar_sdk/src/source/source_packet_ros.hpp:168:30: error: ‘const RslidarPacket’ {aka ‘const struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘header’
  168 |   rs_msg.timestamp = ros_msg.header.stamp.sec + double(ros_msg.header.stamp.nanosec) / 1e9;
      |                              ^~~~~~
/home/x/workspace/src/rslidar_sdk/src/source/source_packet_ros.hpp:168:64: error: ‘const RslidarPacket’ {aka ‘const struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘header’
  168 |   rs_msg.timestamp = ros_msg.header.stamp.sec + double(ros_msg.header.stamp.nanosec) / 1e9;
      |                                                                ^~~~~~
/home/x/workspace/src/rslidar_sdk/src/source/source_packet_ros.hpp:170:29: error: ‘const RslidarPacket’ {aka ‘const struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘is_difop’
  170 |   rs_msg.is_difop = ros_msg.is_difop;
      |                             ^~~~~~~~
/home/x/workspace/src/rslidar_sdk/src/source/source_packet_ros.hpp:171:35: error: ‘const RslidarPacket’ {aka ‘const struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘is_frame_begin’
  171 |   rs_msg.is_frame_begin = ros_msg.is_frame_begin;

TZECHIN6, 2022-10-31:
Same here, got the same error with you. Telling about missing header about `RslidarPacket`

YJXTTZ, 2022-10-31:
They give me a new code, maybe the issue from VMware

TZECHIN6, 2022-10-31:
What do you mean a new code? I am using Docker environment, might be this is the issue?

Now you fix it or not? btw how do you contact them? I cannot find any support from their website.

YJXTTZ, 2022-10-31:
The code that they run in VMware ubuntu20.04 environment successfully, I can contact them because our company brought the lidar.
you can try don't useing virtual environment.

TZECHIN6, 2022-10-31:
I also bought a RS-Ruby Lite. Would you telling the way you contact them? for example email or sth?
I have already built all my ros2 package in Docker already, with extra sensor device like radar gnss camera etc.
it would be a pain if I have to shift to local install just because of this...