RoboSense-LiDAR/rslidar_sdk #101: Colcon build error - ROS2

jclinton830, 2022-09-02:
I am trying to build this package in ubuntu 22.04 with ROS2 Humble. I get the following error log when I run colcon build. Boost and all other deps are already installed. 
```
--- stderr: rslidar_sdk                         
=============================================================
-- ROS Not Found, Ros Support is turned Off!
=============================================================
=============================================================
-- ROS2 Found, Ros2 Support is turned On!
=============================================================
=============================================================
-- Protobuf Found, Protobuf Support is turned On!
=============================================================
=============================================================
-- Cmake run for UNIX GNU Compiler
=============================================================
=============================================================
-- rs_driver Version : v1.3.2
=============================================================
=============================================================
-- POINT_TYPE is XYZI
=============================================================
CMake Error at CMakeLists.txt:150 (add_executable):
  Target "rslidar_sdk_node" links to target "Boost::regex" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< rslidar_sdk [3.84s, exited with code 1]

Summary: 7 packages finished [6.73s]
  1 package failed: rslidar_sdk
  1 package had stderr output: rslidar_sdk
```

Anyone know what I could do to solve it or is this a bug?

skleymenov, 2022-09-14:
I've same issue

jclinton830, 2022-09-19:
> I've same issue

I solved this issue by doing the following. 

First, I cloned, compiled and built the rs_driver source code by doing this 

```
sudo apt-get install libboost-dev libpcap-dev libpcl-dev libeigen3-dev git
mkdir -p ~/robosense_repos && cd ~/robosense_repos
git clone https://github.com/RoboSense-LiDAR/rs_driver.git
mkdir -p rs_driver/build && cd rs_driver/build
cmake ..
make -j4
sudo make install
```

Then I cloned rslidar_sdk and rslidar_msgs repositories into the /src folder of my ROS2 workspace in Ubuntu 22.04.

Then to make rslidar_sdk work on ROS2 Humble I had to modify the CMakelists.txt as follows 

```
@@ -1,209 +1,218 @@
cmake_minimum_required(VERSION 3.5)
cmake_policy(SET CMP0048 NEW)
project(rslidar_sdk)

#=======================================
# Compile setup (ORIGINAL, CATKIN, COLCON)
#=======================================
set(COMPILE_METHOD CATKIN)
set(COMPILE_METHOD COLCON)

#=======================================
# Custom Point Type (XYZI, XYZIRT)
#=======================================
set(POINT_TYPE XYZI)

#========================
# Project details / setup
#========================
set(PROJECT_NAME rslidar_sdk)
add_definitions(-DPROJECT_PATH="${PROJECT_SOURCE_DIR}")
set(CMAKE_BUILD_TYPE RELEASE)
add_definitions(-O3)
add_definitions(-std=c++14)
add_definitions(-std=c++17)
add_compile_options(-Wall)

#========================
# Dependencies Setup
#========================
#ROS#
find_package(roscpp 1.12 QUIET)
if(roscpp_FOUND)
  message(=============================================================)
  message("-- ROS Found, Ros Support is turned On!")
  message(=============================================================)
  configure_file(${CMAKE_CURRENT_SOURCE_DIR}/package_ros1.xml ${CMAKE_CURRENT_SOURCE_DIR}/package.xml COPYONLY)
  add_definitions(-DROS_FOUND)
  configure_file(${CMAKE_CURRENT_SOURCE_DIR}/package_ros1.xml ${CMAKE_CURRENT_SOURCE_DIR}/package.xml COPYONLY)

  find_package(roslib QUIET)
  include_directories(${roscpp_INCLUDE_DIRS} ${roslib_INCLUDE_DIRS})
  set(ROS_LIBS ${roscpp_LIBRARIES} ${roslib_LIBRARIES})

else(roscpp_FOUND)
  message(=============================================================)
  message("-- ROS Not Found, Ros Support is turned Off!")
  message(=============================================================)
endif(roscpp_FOUND)
[... 161 lines cut ...]
                            std_msgs 
                            rslidar_msg 
                            pcl_ros
                            pcl_conversions)
  install(TARGETS
    rslidar_sdk_node
    DESTINATION lib/${PROJECT_NAME}
  )
  install(DIRECTORY
    launch
    rviz
    DESTINATION share/${PROJECT_NAME}
  )
  ament_package()
endif(rclcpp_FOUND AND ${COMPILE_METHOD} STREQUAL "COLCON")

```

Then I had to change one of the import definitions in 

`~/dev_ws/src/rslidar_sdk/src/msg/rs_msg/lidar_point_cloud_msg.h`

by replacing `#include <pcl/io/io.h>` with `#include <pcl/io/pcd_io.h>`

Finally, execute `colcon build` from the root of your ROS2 workspace.