RoboSense-LiDAR/rslidar_sdk #105: Colcon build error in ROS2 galactic desktop docker environment

TZECHIN6, 2022-10-27:
Hi, in short the error message is indicating that there are a missing file `rslidar_msg/msg/rslidar_scan.hpp`. 
Any one has idea how to solve this?

```
Starting >>> rslidar_msg
Finished <<< rslidar_msg [4.85s]                     
Starting >>> rslidar_sdk
--- stderr: rslidar_sdk                             
=============================================================
-- ROS Not Found, Ros Support is turned Off!
=============================================================
=============================================================
-- ROS2 Found, Ros2 Support is turned On!
=============================================================
=============================================================
-- Protobuf Found, Protobuf Support is turned On!
=============================================================
=============================================================
-- Cmake run for UNIX GNU Compiler
=============================================================
=============================================================
-- rs_driver Version : v1.3.2
=============================================================
=============================================================
-- POINT_TYPE is XYZI
=============================================================
In file included from /usr/include/string.h:495,
                 from /usr/include/google/protobuf/stubs/port.h:38,
                 from /usr/include/google/protobuf/stubs/common.h:46,
                 from /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/point_cloud.pb.h:9,
                 from /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/point_cloud.pb.cc:4:
In function ‘void* memset(void*, int, size_t)’,
    inlined from ‘void proto_msg::LidarPointCloud::SharedCtor()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/point_cloud.pb.cc:164:11,
    inlined from ‘proto_msg::LidarPointCloud::LidarPointCloud()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/point_cloud.pb.cc:143:13,
    inlined from ‘void protobuf_point_5fcloud_2eproto::InitDefaultsLidarPointCloud()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/point_cloud.pb.cc:139:1:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:71:33: warning: ‘void* __builtin_memset(void*, int, long unsigned int)’ offset [57, 69] from the object at ‘proto_msg::_LidarPointCloud_default_instance_’ is out of the bounds of referenced subobject ‘proto_msg::LidarPointCloud::timestamp_’ with type ‘double’ at offset 48 [-Warray-bounds]
   71 |   return __builtin___memset_chk (__dest, __ch, __len, __bos0 (__dest));
      |          ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/string.h:495,
                 from /usr/include/google/protobuf/stubs/port.h:38,
                 from /usr/include/google/protobuf/stubs/common.h:46,
                 from /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/scan.pb.h:9,
                 from /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/scan.pb.cc:4:
In function ‘void* memset(void*, int, size_t)’,
    inlined from ‘void proto_msg::LidarScan::SharedCtor()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/scan.pb.cc:145:11,
    inlined from ‘proto_msg::LidarScan::LidarScan()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/scan.pb.cc:129:13,
    inlined from ‘void protobuf_scan_2eproto::InitDefaultsLidarScan()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/scan.pb.cc:125:1:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:71:33: warning: ‘void* __builtin_memset(void*, int, long unsigned int)’ offset [57, 60] from the object at ‘proto_msg::_LidarScan_default_instance_’ is out of the bounds of referenced subobject ‘proto_msg::LidarScan::timestamp_’ with type ‘double’ at offset 48 [-Warray-bounds]
   71 |   return __builtin___memset_chk (__dest, __ch, __len, __bos0 (__dest));
      |          ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /robosense_ws/src/rslidar_sdk/src/adapter/packet_ros_adapter.hpp:139,
                 from /robosense_ws/src/rslidar_sdk/src/manager/adapter_manager.h:75,
                 from /robosense_ws/src/rslidar_sdk/node/rslidar_sdk_node.cpp:34:
/robosense_ws/src/rslidar_sdk/src/msg/ros_msg_translator.h:144:10: fatal error: rslidar_msg/msg/rslidar_scan.hpp: No such file or directory
  144 | #include "rslidar_msg/msg/rslidar_scan.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /robosense_ws/src/rslidar_sdk/src/adapter/packet_ros_adapter.hpp:139,
                 from /robosense_ws/src/rslidar_sdk/src/manager/adapter_manager.h:75,
                 from /robosense_ws/src/rslidar_sdk/src/manager/adapter_manager.cpp:33:
/robosense_ws/src/rslidar_sdk/src/msg/ros_msg_translator.h:144:10: fatal error: rslidar_msg/msg/rslidar_scan.hpp: No such file or directory
  144 | #include "rslidar_msg/msg/rslidar_scan.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
make[2]: *** [CMakeFiles/rslidar_sdk_node.dir/build.make:63: CMakeFiles/rslidar_sdk_node.dir/node/rslidar_sdk_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/rslidar_sdk_node.dir/build.make:76: CMakeFiles/rslidar_sdk_node.dir/src/manager/adapter_manager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:98: CMakeFiles/rslidar_sdk_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rslidar_sdk [4.21s, exited with code 2]

Summary: 1 package finished [9.23s]
  1 package failed: rslidar_sdk
  1 package had stderr output: rslidar_sdk
Completed building the worksapce.
```

TZECHIN6, 2022-10-27:
Even in the official repos https://github.com/RoboSense-LiDAR/rslidar_msg
There is no file called `rslidar_scan.hpp` under msg directory....

So is that a bug? Do/Should i have to delete this line to make thing works?

enginbaglayici, 2022-10-28:
I am having the same issue.. Looks like the msg definition is not complete. Have you found a workaround? @TZECHIN6

TZECHIN6, 2022-10-28:
@enginbaglayici how can you find it, amazing! I will reinstall it on monday.

I will keep this be opened for a while, and see whether the developer will answer it

TZECHIN6, 2022-10-31:
@enginbaglayici  I am still facing the same error, would you please explain more about that? do you copy those lines into the `RslidarPacket.msg`? instead of adding the miss .msg file?

TZECHIN6, 2022-10-31:
I have added the new line `RslidarPacket[] packets` into the `RslidarPacket.msg` .msg file. The missing error is now gone. However now issue come up:
```
Starting >>> rslidar_msg
--- stderr: rslidar_msg                             
In file included from /robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.h:19,
                 from /robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c:4:
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__struct.h:35:3: error: unknown type name ‘rslidar_msg__msg__RslidarPacket__Sequence’
   35 |   rslidar_msg__msg__RslidarPacket__Sequence packets;
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c: In function ‘rslidar_msg__msg__RslidarPacket__init’:
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c:40:56: warning: passing argument 1 of ‘rslidar_msg__msg__RslidarPacket__Sequence__init’ from incompatible pointer type [-Wincompatible-pointer-types]
   40 |   if (!rslidar_msg__msg__RslidarPacket__Sequence__init(&msg->packets, 0)) {
      |                                                        ^~~~~~~~~~~~~
      |                                                        |
      |                                                        int *
In file included from /robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c:4:
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.h:109:93: note: expected ‘rslidar_msg__msg__RslidarPacket__Sequence *’ {aka ‘struct rslidar_msg__msg__RslidarPacket__Sequence *’} but argument is of type ‘int *’
  109 | rslidar_msg__msg__RslidarPacket__Sequence__init(rslidar_msg__msg__RslidarPacket__Sequence * array, size_t size);
      |                                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c: In function ‘rslidar_msg__msg__RslidarPacket__fini’:
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c:60:51: warning: passing argument 1 of ‘rslidar_msg__msg__RslidarPacket__Sequence__fini’ from incompatible pointer type [-Wincompatible-pointer-types]
   60 |   rslidar_msg__msg__RslidarPacket__Sequence__fini(&msg->packets);
      |                                                   ^~~~~~~~~~~~~
      |                                                   |
      |                                                   int *
In file included from /robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c:4:
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.h:121:93: note: expected ‘rslidar_msg__msg__RslidarPacket__Sequence *’ {aka ‘struct rslidar_msg__msg__RslidarPacket__Sequence *’} but argument is of type ‘int *’
  121 | rslidar_msg__msg__RslidarPacket__Sequence__fini(rslidar_msg__msg__RslidarPacket__Sequence * array);
      |                                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c: In function ‘rslidar_msg__msg__RslidarPacket__are_equal’:
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c:91:7: warning: passing argument 1 of ‘rslidar_msg__msg__RslidarPacket__Sequence__are_equal’ from incompatible pointer type [-Wincompatible-pointer-types]
   91 |       &(lhs->packets), &(rhs->packets)))
      |       ^~~~~~~~~~~~~~~
      |       |
      |       const int *
In file included from /robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c:4:
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.h:154:104: note: expected ‘const rslidar_msg__msg__RslidarPacket__Sequence *’ {aka ‘const struct rslidar_msg__msg__RslidarPacket__Sequence *’} but argument is of type ‘const int *’
  154 | rslidar_msg__msg__RslidarPacket__Sequence__are_equal(const rslidar_msg__msg__RslidarPacket__Sequence * lhs, const rslidar_msg__msg__RslidarPacket__Sequence * rhs);
      |                                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c:91:24: warning: passing argument 2 of ‘rslidar_msg__msg__RslidarPacket__Sequence__are_equal’ from incompatible pointer type [-Wincompatible-pointer-types]
   91 |       &(lhs->packets), &(rhs->packets)))
      |                        ^~~~~~~~~~~~~~~
      |                        |
      |                        const int *
In file included from /robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.c:4:
/robosense_ws/build/rslidar_msg/rosidl_generator_c/rslidar_msg/msg/detail/rslidar_packet__functions.h:154:159: note: expected ‘const rslidar_msg__msg__RslidarPacket__Sequence *’ {aka ‘const struct rslidar_msg__msg__RslidarPacket__Sequence *’} but argument is of type ‘const int *’
[... cut ...]
```
/robosense_ws/build/rslidar_msg/rosidl_typesupport_introspection_cpp/rslidar_msg/msg/detail/rslidar_packet__type_support.cpp:201:66: error: specialization of ‘const rosidl_message_type_support_t* rosidl_typesupport_introspection_cpp::get_message_type_support_handle() [with T = rslidar_msg::msg::RslidarPacket_<std::allocator<void> >; rosidl_message_type_support_t = rosidl_message_type_support_t]’ after instantiation
  201 | get_message_type_support_handle<rslidar_msg::msg::RslidarPacket>()
```

TZECHIN6, 2022-10-31:
I retried without using the above build.script, instead straightly follow the guide provided in repo.

The error is telling that missing header of `RslidarPacket`

```
In file included from /robosense_ws/src/rslidar_sdk/src/manager/node_manager.cpp:36:
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp: In function ‘robosense::lidar::Packet robosense::lidar::toRsMsg(const RslidarPacket&)’:
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp:230:30: error: ‘const RslidarPacket’ {aka ‘const struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘header’
  230 |   rs_msg.timestamp = ros_msg.header.stamp.sec + double(ros_msg.header.stamp.nanosec) / 1e9;
      |                              ^~~~~~
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp:230:64: error: ‘const RslidarPacket’ {aka ‘const struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘header’
  230 |   rs_msg.timestamp = ros_msg.header.stamp.sec + double(ros_msg.header.stamp.nanosec) / 1e9;
      |                                                                ^~~~~~
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp:232:29: error: ‘const RslidarPacket’ {aka ‘const struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘is_difop’
  232 |   rs_msg.is_difop = ros_msg.is_difop;
      |                             ^~~~~~~~
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp:233:35: error: ‘const RslidarPacket’ {aka ‘const struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘is_frame_begin’
  233 |   rs_msg.is_frame_begin = ros_msg.is_frame_begin;
      |                                   ^~~~~~~~~~~~~~
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp: In function ‘rslidar_msg::msg::RslidarPacket robosense::lidar::toRosMsg(const robosense::lidar::Packet&, const string&)’:
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp:289:11: error: ‘using RslidarPacket = struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’ {aka ‘struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘header’
  289 |   ros_msg.header.stamp.sec = (uint32_t)floor(rs_msg.timestamp);
      |           ^~~~~~
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp:290:11: error: ‘using RslidarPacket = struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’ {aka ‘struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘header’
  290 |   ros_msg.header.stamp.nanosec = (uint32_t)round((rs_msg.timestamp - ros_msg.header.stamp.sec) * 1e9);
      |           ^~~~~~
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp:290:78: error: ‘using RslidarPacket = struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’ {aka ‘struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘header’
  290 |   ros_msg.header.stamp.nanosec = (uint32_t)round((rs_msg.timestamp - ros_msg.header.stamp.sec) * 1e9);
      |                                                                              ^~~~~~
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp:292:11: error: ‘using RslidarPacket = struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’ {aka ‘struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘header’
  292 |   ros_msg.header.frame_id = frame_id;
      |           ^~~~~~
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp:293:11: error: ‘using RslidarPacket = struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’ {aka ‘struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘is_difop’
  293 |   ros_msg.is_difop = rs_msg.is_difop;
      |           ^~~~~~~~
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp:294:11: error: ‘using RslidarPacket = struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’ {aka ‘struct rslidar_msg::msg::RslidarPacket_<std::allocator<void> >’} has no member named ‘is_frame_begin’
  294 |   ros_msg.is_frame_begin = rs_msg.is_frame_begin;
      |           ^~~~~~~~~~~~~~
/robosense_ws/src/rslidar_sdk/src/source/source_packet_ros.hpp:298:18: error: ‘using _data_type = struct std::array<unsigned char, 1248>’ {aka ‘struct std::array<unsigned char, 1248>’} has no member named ‘emplace_back’
  298 |     ros_msg.data.emplace_back(rs_msg.buf_[i]);
      |                  ^~~~~~~~~~~~
make[2]: *** [CMakeFiles/rslidar_sdk_node.dir/build.make:76: CMakeFiles/rslidar_sdk_node.dir/src/manager/node_manager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:98: CMakeFiles/rslidar_sdk_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rslidar_sdk [5.90s, exited with code 2]

Summary: 1 package finished [10.6s]
  1 package failed: rslidar_sdk
  1 package had stderr output: rslidar_sdk
```

enginbaglayici, 2022-10-31:
I haven't touched the msg definitions but directly copied the [rslidar_msg dev branch ](https://github.com/RoboSense-LiDAR/rslidar_msg/blob/dev/msg/RslidarPacket.msg), made a clean build and it worked without any of the above issues you faced.

TZECHIN6, 2022-10-31:
Just tried, no luck.... What environment are you using for the build?

TZECHIN6, 2022-10-31:
Might be its because costed by ROS2 galactic? not sure... they said it supports.
Have you tried to install in Docker environment? for example docker pull a osrf/ros:galactic-desktop, then run it and build workspace as usual.

enginbaglayici, 2022-10-31:
I don't think the problem is caused by the ros version as well.
I also built the packages in an NVIDIA Jetson container which is [dustynv/ros:foxy-ros-base-l4t-r32.5.0](https://github.com/dusty-nv/jetson-containers/blob/master/README.md) but didn't try the one you mentioned.

TZECHIN6, 2022-11-01:
Are you using Git to download the SDK or using the zip file?

RS-Alex, 2022-11-04:
> Hi, in short the error message is indicating that there are a missing file `rslidar_msg/msg/rslidar_scan.hpp`. Any one has idea how to solve this?
> 
> The folder structure should be correct
> 
> ```
> .
> ├── build-workspace-general.sh
> └── src
>     ├── rslidar_msg
>     └── rslidar_sdk
> ```
> 
> Error message
> 
> ```
> Starting >>> rslidar_msg
> Finished <<< rslidar_msg [4.85s]                     
> Starting >>> rslidar_sdk
> --- stderr: rslidar_sdk                             
> =============================================================
> -- ROS Not Found, Ros Support is turned Off!
> =============================================================
> =============================================================
> -- ROS2 Found, Ros2 Support is turned On!
> =============================================================
> =============================================================
> -- Protobuf Found, Protobuf Support is turned On!
> =============================================================
> =============================================================
> -- Cmake run for UNIX GNU Compiler
> =============================================================
> =============================================================
> -- rs_driver Version : v1.3.2
> =============================================================
> =============================================================
> -- POINT_TYPE is XYZI
> =============================================================
> In file included from /usr/include/string.h:495,
>                  from /usr/include/google/protobuf/stubs/port.h:38,
>                  from /usr/include/google/protobuf/stubs/common.h:46,
>                  from /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/point_cloud.pb.h:9,
>                  from /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/point_cloud.pb.cc:4:
> In function ‘void* memset(void*, int, size_t)’,
>     inlined from ‘void proto_msg::LidarPointCloud::SharedCtor()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/point_cloud.pb.cc:164:11,
>     inlined from ‘proto_msg::LidarPointCloud::LidarPointCloud()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/point_cloud.pb.cc:143:13,
>     inlined from ‘void protobuf_point_5fcloud_2eproto::InitDefaultsLidarPointCloud()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/point_cloud.pb.cc:139:1:
> /usr/include/x86_64-linux-gnu/bits/string_fortified.h:71:33: warning: ‘void* __builtin_memset(void*, int, long unsigned int)’ offset [57, 69] from the object at ‘proto_msg::_LidarPointCloud_default_instance_’ is out of the bounds of referenced subobject ‘proto_msg::LidarPointCloud::timestamp_’ with type ‘double’ at offset 48 [-Warray-bounds]
>    71 |   return __builtin___memset_chk (__dest, __ch, __len, __bos0 (__dest));
>       |          ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> In file included from /usr/include/string.h:495,
>                  from /usr/include/google/protobuf/stubs/port.h:38,
>                  from /usr/include/google/protobuf/stubs/common.h:46,
>                  from /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/scan.pb.h:9,
>                  from /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/scan.pb.cc:4:
> In function ‘void* memset(void*, int, size_t)’,
>     inlined from ‘void proto_msg::LidarScan::SharedCtor()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/scan.pb.cc:145:11,
>     inlined from ‘proto_msg::LidarScan::LidarScan()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/scan.pb.cc:129:13,
>     inlined from ‘void protobuf_scan_2eproto::InitDefaultsLidarScan()’ at /robosense_ws/src/rslidar_sdk/src/msg/proto_msg/scan.pb.cc:125:1:
> /usr/include/x86_64-linux-gnu/bits/string_fortified.h:71:33: warning: ‘void* __builtin_memset(void*, int, long unsigned int)’ offset [57, 60] from the object at ‘proto_msg::_LidarScan_default_instance_’ is out of the bounds of referenced subobject ‘proto_msg::LidarScan::timestamp_’ with type ‘double’ at offset 48 [-Warray-bounds]
>    71 |   return __builtin___memset_chk (__dest, __ch, __len, __bos0 (__dest));
>       |          ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> In file included from /robosense_ws/src/rslidar_sdk/src/adapter/packet_ros_adapter.hpp:139,
>                  from /robosense_ws/src/rslidar_sdk/src/manager/adapter_manager.h:75,
>                  from /robosense_ws/src/rslidar_sdk/node/rslidar_sdk_node.cpp:34:
> /robosense_ws/src/rslidar_sdk/src/msg/ros_msg_translator.h:144:10: fatal error: rslidar_msg/msg/rslidar_scan.hpp: No such file or directory
>   144 | #include "rslidar_msg/msg/rslidar_scan.hpp"
>       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> In file included from /robosense_ws/src/rslidar_sdk/src/adapter/packet_ros_adapter.hpp:139,
>                  from /robosense_ws/src/rslidar_sdk/src/manager/adapter_manager.h:75,
>                  from /robosense_ws/src/rslidar_sdk/src/manager/adapter_manager.cpp:33:
> /robosense_ws/src/rslidar_sdk/src/msg/ros_msg_translator.h:144:10: fatal error: rslidar_msg/msg/rslidar_scan.hpp: No such file or directory
>   144 | #include "rslidar_msg/msg/rslidar_scan.hpp"
>       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> compilation terminated.
> compilation terminated.
> make[2]: *** [CMakeFiles/rslidar_sdk_node.dir/build.make:63: CMakeFiles/rslidar_sdk_node.dir/node/rslidar_sdk_node.cpp.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
[... cut ...]
> make[1]: *** [CMakeFiles/Makefile2:98: CMakeFiles/rslidar_sdk_node.dir/all] Error 2
> make: *** [Makefile:141: all] Error 2 
> ---
> Failed   <<< rslidar_sdk [4.21s, exited with code 2]
> 
> Summary: 1 package finished [9.23s]
>   1 package failed: rslidar_sdk
>   1 package had stderr output: rslidar_sdk
> Completed building the worksapce.
> ```

Hi @TZECHIN6 we have located and fixed this issue, please download both RS-SDK and RS-driver again from the "release" branch and re-compile 
[image]

TZECHIN6, 2022-11-04:
@RS-Alex  thanks for your investigation. Do you mean rslidar_msg? instead of rs_driver? as the rslidar_sdk should include the drive according to the README

TZECHIN6, 2022-11-04:
@RS-Alex I used the release branch of rslidar_msh and rslidar_sdk to compile the workspace.

```
root@thomasluk-DesktopPC:/robosense_ws# colcon build --symlink-install
Starting >>> rslidar_msg
Finished <<< rslidar_msg [4.44s]                     
Starting >>> rslidar_sdk
--- stderr: rslidar_sdk                                
=============================================================
-- POINT_TYPE is XYZI
=============================================================
=============================================================
-- ROS Not Found. ROS Support is turned Off.
=============================================================
=============================================================
-- ROS2 Found. ROS2 Support is turned On.
=============================================================
=============================================================
-- CMake run for UNIX GNU Compiler
=============================================================
=============================================================
-- rs_driver Version : v1.5.7
=============================================================
---
Finished <<< rslidar_sdk [15.8s]

Summary: 2 packages finished [20.4s]
  1 package had stderr output: rslidar_sdk
```

The workspace seem compile successfully. Thanks. However, I got issue on running the node:

```
root@thomasluk-DesktopPC:/robosense_ws# ros2 launch rslidar_sdk start.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-11-04-06-20-36-706642-thomasluk-DesktopPC-1732
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rslidar_sdk_node-1]: process started with pid [1734]
[INFO] [rviz2-2]: process started with pid [1736]
[rslidar_sdk_node-1] ********************************************************
[rslidar_sdk_node-1] **********                                    **********
[rslidar_sdk_node-1] **********    RSLidar_SDK Version: v1.5.7     **********
[rslidar_sdk_node-1] **********                                    **********
[rslidar_sdk_node-1] ********************************************************
[rslidar_sdk_node-1] ------------------------------------------------------
[rslidar_sdk_node-1] Receive Packets From : Pcap
[rslidar_sdk_node-1] Msop Port: 1990
[rslidar_sdk_node-1] Difop Port: 1991
[rslidar_sdk_node-1] ------------------------------------------------------
[rslidar_sdk_node-1] ------------------------------------------------------
[rslidar_sdk_node-1]              RoboSense Driver Parameters 
[rslidar_sdk_node-1] input type: PCAP_FILE
[rslidar_sdk_node-1] lidar_type: RSM1
[rslidar_sdk_node-1] ------------------------------------------------------
[rslidar_sdk_node-1] ------------------------------------------------------
[rslidar_sdk_node-1]              RoboSense Input Parameters 
[rslidar_sdk_node-1] msop_port: 1990
[rslidar_sdk_node-1] difop_port: 1991
[rslidar_sdk_node-1] host_address: 0.0.0.0
[rslidar_sdk_node-1] group_address: 0.0.0.0
[rslidar_sdk_node-1] pcap_path: /mnt/shared/pcap/M1/m1_1990_1991.pcap
[rslidar_sdk_node-1] pcap_rate: 1
[rslidar_sdk_node-1] pcap_repeat: 1
[rslidar_sdk_node-1] use_vlan: 0
[rslidar_sdk_node-1] user_layer_bytes: 0
[rslidar_sdk_node-1] tail_layer_bytes: 0
[rslidar_sdk_node-1] ------------------------------------------------------
[rslidar_sdk_node-1] ------------------------------------------------------
[rslidar_sdk_node-1]              RoboSense Decoder Parameters 
[rslidar_sdk_node-1] wait_for_difop: 1
[rslidar_sdk_node-1] min_distance: 0.2
[rslidar_sdk_node-1] max_distance: 200
[rslidar_sdk_node-1] start_angle: 0
[rslidar_sdk_node-1] end_angle: 360
[rslidar_sdk_node-1] use_lidar_clock: 0
[rslidar_sdk_node-1] dense_points: 0
[rslidar_sdk_node-1] config_from_file: 0
[rslidar_sdk_node-1] angle_path: 
[rslidar_sdk_node-1] split_frame_mode: 1
[rslidar_sdk_node-1] split_angle: 0
[rslidar_sdk_node-1] num_blks_split: 0
[rslidar_sdk_node-1] ------------------------------------------------------
[rslidar_sdk_node-1] ------------------------------------------------------
[rslidar_sdk_node-1]              RoboSense Transform Parameters 
[rslidar_sdk_node-1] x: 0
[rslidar_sdk_node-1] y: 0
[rslidar_sdk_node-1] z: 0
[rslidar_sdk_node-1] roll: 0
[rslidar_sdk_node-1] pitch: 0
[rslidar_sdk_node-1] yaw: 0
[rslidar_sdk_node-1] ------------------------------------------------------
[rslidar_sdk_node-1] ERRCODE_PCAPWRONGPATH
[rslidar_sdk_node-1] Driver Initialize Error....
[ERROR] [rslidar_sdk_node-1]: process has died [pid 1734, exit code 255, cmd '/robosense_ws/install/rslidar_sdk/lib/rslidar_sdk/rslidar_sdk_node --ros-args -r __ns:=/rslidar_sdk'].
[rviz2-2] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[rviz2-2] [INFO] [1667542836.928622685] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1667542836.928686773] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[rviz2-2] [INFO] [1667542836.948627768] [rviz2]: Stereo is NOT SUPPORTED
```

[rslidar_sdk_node-1] ERRCODE_PCAPWRONGPATH
[rslidar_sdk_node-1] Driver Initialize Error....

Above statement said the driver got some issues... so I am not sure the build is success or not.

i am running on a docker environment with --network host flag, it a ubuntu20.04 base ROS2 galactic desktop image.

RS-Alex, 2022-11-04:
> @RS-Alex I used the release branch of rslidar_msh and rslidar_sdk to compile the workspace.
> 
> ```
> root@thomasluk-DesktopPC:/robosense_ws# colcon build --symlink-install
> Starting >>> rslidar_msg
> Finished <<< rslidar_msg [4.44s]                     
> Starting >>> rslidar_sdk
> --- stderr: rslidar_sdk                                
> =============================================================
> -- POINT_TYPE is XYZI
> =============================================================
> =============================================================
> -- ROS Not Found. ROS Support is turned Off.
> =============================================================
> =============================================================
> -- ROS2 Found. ROS2 Support is turned On.
> =============================================================
> =============================================================
> -- CMake run for UNIX GNU Compiler
> =============================================================
> =============================================================
> -- rs_driver Version : v1.5.7
> =============================================================
> ---
> Finished <<< rslidar_sdk [15.8s]
> 
> Summary: 2 packages finished [20.4s]
>   1 package had stderr output: rslidar_sdk
> ```
> 
> The workspace seem compile successfully. Thanks. However, I got issue on running the node:
> 
> ```
> root@thomasluk-DesktopPC:/robosense_ws# ros2 launch rslidar_sdk start.py
> [INFO] [launch]: All log files can be found below /root/.ros/log/2022-11-04-06-45-18-676955-thomasluk-DesktopPC-1779
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [rslidar_sdk_node-1]: process started with pid [1781]
> [INFO] [rviz2-2]: process started with pid [1783]
> [rslidar_sdk_node-1] ********************************************************
> [rslidar_sdk_node-1] **********                                    **********
> [rslidar_sdk_node-1] **********    RSLidar_SDK Version: v1.5.7     **********
> [rslidar_sdk_node-1] **********                                    **********
> [rslidar_sdk_node-1] ********************************************************
> [rslidar_sdk_node-1] ------------------------------------------------------
> [rslidar_sdk_node-1] Receive Packets From : Online LiDAR
> [rslidar_sdk_node-1] Msop Port: 6699
> [rslidar_sdk_node-1] Difop Port: 7788
> [rslidar_sdk_node-1] ------------------------------------------------------
> [rslidar_sdk_node-1] ------------------------------------------------------
> [rslidar_sdk_node-1]              RoboSense Driver Parameters 
> [rslidar_sdk_node-1] input type: ONLINE_LIDAR
> [rslidar_sdk_node-1] lidar_type: RS80
> [rslidar_sdk_node-1] ------------------------------------------------------
> [rslidar_sdk_node-1] ------------------------------------------------------
> [rslidar_sdk_node-1]              RoboSense Input Parameters 
> [rslidar_sdk_node-1] msop_port: 6699
> [rslidar_sdk_node-1] difop_port: 7788
> [rslidar_sdk_node-1] host_address: 0.0.0.0
> [rslidar_sdk_node-1] group_address: 0.0.0.0
> [rslidar_sdk_node-1] pcap_path: /mnt/shared/pcap/M1/m1_1990_1991.pcap
> [rslidar_sdk_node-1] pcap_rate: 1
> [rslidar_sdk_node-1] pcap_repeat: 1
> [rslidar_sdk_node-1] use_vlan: 0
> [rslidar_sdk_node-1] user_layer_bytes: 0
> [rslidar_sdk_node-1] tail_layer_bytes: 0
> [rslidar_sdk_node-1] ------------------------------------------------------
> [rslidar_sdk_node-1] ------------------------------------------------------
> [rslidar_sdk_node-1]              RoboSense Decoder Parameters 
> [rslidar_sdk_node-1] wait_for_difop: 1
> [rslidar_sdk_node-1] min_distance: 0.2
[... 40 lines cut ...]
> [rslidar_sdk_node-1] ERRCODE_MSOPTIMEOUT
> [rslidar_sdk_node-1] ERRCODE_MSOPTIMEOUT
> [rslidar_sdk_node-1] ERRCODE_MSOPTIMEOUT
> [rslidar_sdk_node-1] ERRCODE_MSOPTIMEOUT
> [rslidar_sdk_node-1] ERRCODE_MSOPTIMEOUT
> [rslidar_sdk_node-1] ERRCODE_MSOPTIMEOUT
> [rslidar_sdk_node-1] ERRCODE_MSOPTIMEOUT
> [rslidar_sdk_node-1] ERRCODE_MSOPTIMEOUT
> [rslidar_sdk_node-1] ERRCODE_MSOPTIMEOUT
> [rslidar_sdk_node-1] ERRCODE_MSOPTIMEOUT
> ^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
> [rviz2-2] [INFO] [1667544340.464576138] [rclcpp]: signal_handler(signal_value=2)
> [rslidar_sdk_node-1] [INFO] [1667544340.464578704] [rclcpp]: signal_handler(signal_value=2)
> [rslidar_sdk_node-1] RoboSense-LiDAR-Driver is stopping.....
> [rslidar_sdk_node-1] ERRCODE_MSOPTIMEOUT
> ```
> 
> I have read the config guideline and should be correct, the host can ping the lidar. No idea how to fix this timeout issue. Is it because due to the GPS issue? I havent insert with the GPS antenna.

Your SDK has started successfully. The MSOPTIMEOUT code indicates that your computer is not receiving data stream from your lidar. You can check the following items:

1. Check if you have selected the correct lidar type in the config file
2. If the port numbers in the config file is correct
3. Check if your computer's IP has been changed correspond to lidar's destination IP (default IP is 192.168.1.102)
4. If you are still not getting the error code, please install Wireshark and check if you can get the UDP data stream normally.

TZECHIN6, 2022-11-04:
@RS-Alex  I have done all the 4-steps you have mentioned. I can see the packet in Wireshark as well...

I am not sure is it caused by my docker environment issue? I run the demo talker and listener node and no receive from listener node https://docs.ros.org/en/galactic/How-To-Guides/Run-2-nodes-in-single-or-separate-docker-containers.html

Any other way/command to check the network issue in docker?

RS-Alex, 2022-11-04:
@TZECHIN6 Please make sure your firewall is closed, and please send the screen shot of the config.yaml file.
May I know what lidar type you are using?

TZECHIN6, 2022-11-04:
I am using RS80, the firewall is closed. 

The problem might be caused by flooded UDP in switch. Power cycles it will solve the issue. at least in my case. Thanks for your help. I will open another issue ticket for new issue.

closed 2022-11-04 by TZECHIN6

lonlonago, 2023-09-06:
port num is error, reset it in config file