RoboSense-LiDAR/rslidar_sdk #112: can't play the ros2 bag with topic rslidar_packets

KusabiSun, 2023-01-19:
Hi,
I have a problem to record and play the ros2 bag for robosense M1. I update the rslidar sdk and rslidar msg to the version 1.5.8.
And I follow the official Dokument to set config.yaml file to record and play the ros2 bag and I took care of the indentation of yaml file. I can record the ros2 bag, but cant play it. The ros2bag info is this:

**Files:             rosbag2_2023_01_19-23_12_50_0.db3
Bag size:          190.8 MiB
Storage id:        sqlite3
Duration:          22.875s
Topic information: Topic: /rslidar_packets | Type: rslidar_msg/msg/RslidarPacket | Count: 144110 | Serialization Format: cdr**

When I run the rslidar_sdk and want to see the point cloud on rivz2. I can't see the cloud point on rivz2. But if i record and play the topic /rslidar_point and change no setting, i can see the point cloud on rivz2. I am not sure what I forget to set and why happens this problem. I can't find similar Problem in Internet and the most case is my setting is somewhere wrong.

Thanks for any replies.
Kusabi

YitianShi, 2023-01-20:
I got the same problem either. Could anyone give a solution?

Ming45, 2023-04-19:
Because `**spin**` is not called to process the callback queue, the callback function is not fired and the packets cannot be resolved.       
Add  **rclcpp::spin(noed)**  after initialization. (You have to get the nodes yourself)

KusabiSun, 2023-04-28:
Thank you for your answer. We find another way to solve this problem.
But thank you very much. 
I will close this issue. If anyone has questions, reopen it again

atauzki, 2023-07-25:
it doesn't receive any packet at all. but calling `rclcpp::spin()` in the packet subscriber node and publisher node just make subscriber work and make the overflow. How can I solve this problem?