RoboSense-LiDAR/rslidar_sdk #116: Point cloud dimensions seems not consistent with real world distances

howaix1, 2023-03-23:
The distance of point cloud measured in rviz seems to be half of the corresponding real world distance.

I tested the ros_rslidar, rslidar_sdk and RSView and this bug only occurred in rslidar_sdk.

This occurred both in XYZI and XYZIRT settings.

Teemumokkonen, 2024-03-19:
To anybody who is still facing this issue: Some RS-lidars seem to assume a 0.01 meter distance resolution. The driver assumes 0.05m resolution, which causes this issue. In the driver decoder implementation change the following line definition of the resolution:
https://github.com/RoboSense-LiDAR/rs_driver/blob/ef1bdd45ee4df51e5acb57b64e49d8fed233039c/src/rs_driver/driver/decoder/decoder_RS16.hpp#L151

Remember to change the decoder version of the lidar that you are using. The link is pointing to the RS16 decoder.

Change the value from 0.005f to 0.01f. This should resolve the issue with the scaling, this change at least for us fixed this real world distance scaling issue. 

Hope this helps anybody who is looking into this same issue.

johnny555, 2025-06-07:
Thanks for this issue. I ran into this problem yesterday and this solved it.
Can you do a pull request for this? 

I suspect I won't have been the only one to run into this issue.