RoboSense-LiDAR/rslidar_sdk #125: ROS2 - Colcon build Error

nicolal01, 2023-06-19:
I am trying to build this package in ubuntu 22.04 with ROS2 Humble. I get the following error log when I run colcon build.
[image]

Does anyone know what i could do to solve this error?

AlessioGianu, 2023-10-05:
hi, i've followed the guideline in the readme but the colcon build command shows the follow:
[image]

How can i solve these errors? Thank you so much

TZECHIN6, 2023-10-06:
might be you provide more details about your build step? seem you miss to source `/opt/ros/<distro>/setup.bash`

AlessioGianu, 2023-10-06:
Hi, thank you for answering;
the source /opt/ros/<distro>/setup.bash is present inside the bashrc file. 
 I followed the following steps:
- Install via git clone
- Install all the dependency shown in the readme
- followed the sub-steps in the [4.3 Compile with ROS2 colcon]

I'm not expert of ROS2 so maybe I'm missing some trivial steps. 
I also tried chatGPT but it wasn't helpful ( people are better ).

sbarber-dsi, 2023-10-06:
I believe you have not configured the project to build for ROS2.

The Robosense ROS driver does not compile for ROS2 as-is.
You must make a number of minor code changes in order for it to compile.

AlessioGianu, 2023-10-06:
Sorry, do you mean changes that are not inside the readMe file?

TZECHIN6, 2023-10-06:
Would you share your workspace structure with `tree` cmd? Also, could you run the demo_node example in ROS2?

I have built it several times today... all built are success.

AlessioGianu, 2023-10-09:
Hi sorry for my late response, i have not my pc with me this WE.
I'm working with ROS2 Humble with full desktop installation and ubuntu 22.04 LTS.

 I run the command `rosdep install --from-paths src --ignore src -r -y` inside the /robosense_ws/src/rslidar_sdk folder but didn't work so instead i run `rosdep install --from-paths src -y -r` and the result was "all deps are installed" 

I want to thank you for your time.

The Tree structure of the ws is:
.
└── src
    ├── rslidar_msg
    │   ├── CMakeLists.txt
    │   ├── LICENSE
    │   ├── msg
    │   │   └── RslidarPacket.msg
    │   ├── package.xml
    │   ├── README.md
    │   ├── ros1
    │   │   ├── CMakeLists.txt
    │   │   └── package.xml
    │   └── ros2
    │       ├── CMakeLists.txt
    │       └── package.xml
    └── rslidar_sdk
        ├── CHANGELOG.md
        ├── CMakeLists.txt
        ├── config
        │   └── config.yaml
        ├── create_debian.sh
        ├── doc
        │   ├── howto
        │   │   ├── 04_how_to_work_along_with_rslidar_sdk_node_v1.3.x_CN.md
        │   │   ├── 05_how_to_change_point_type_CN.md
        │   │   ├── 05_how_to_change_point_type.md
        │   │   ├── 06_how_to_decode_online_lidar_CN.md
        │   │   ├── 06_how_to_decode_online_lidar.md
        │   │   ├── 07_online_lidar_advanced_topics_CN.md
        │   │   ├── 07_online_lidar_advanced_topics.md
        │   │   ├── 08_how_to_decode_pcap_file_CN.md
        │   │   ├── 08_how_to_decode_pcap_file.md
        │   │   ├── 09_pcap_file_advanced_topics_CN.md
        │   │   ├── 09_pcap_file_advanced_topics.md
        │   │   ├── 10_how_to_use_coordinate_transformation_CN.md
        │   │   ├── 10_how_to_use_coordinate_transformation.md
        │   │   ├── 11_how_to_record_replay_packet_rosbag_CN.md
        │   │   ├── 11_how_to_record_replay_packet_rosbag.md
        │   │   ├── 12_how_to_create_deb.md
        │   │   └── img
        │   │       ├── 04_01_packet_rosbag.png
        │   │       ├── 07_01_broadcast.png
        │   │       ├── 07_02_unicast.png
        │   │       ├── 07_03_multicast.png
        │   │       ├── 07_04_multi_lidars_port.png
        │   │       ├── 07_05_multi_lidars_ip.png
        │   │       ├── 07_06_vlan_layer.png
        │   │       ├── 07_07_vlan.png
        │   │       └── 07_08_user_layer.png
        │   ├── intro
        │   │   ├── 02_parameter_intro_CN.md
        │   │   ├── 02_parameter_intro.md
        │   │   ├── 03_hiding_parameters_intro_CN.md
        │   │   └── 03_hiding_parameters_intro.md
        │   └── src_intro
        │       ├── img
        │       │   ├── class_destination_packet.png
        │       │   ├── class_destination_pointcloud.png
        │       │   ├── class_node_manager.png
        │       │   ├── class_source_destination.png
        │       │   ├── class_source_driver.png
        │       │   └── class_source_packet_ros.png
[... 267 lines cut ...]
            │   │   ├── split_strategy_test.cpp
            │   │   ├── sync_queue_test.cpp
            │   │   └── trigon_test.cpp
            │   ├── tool
            │   │   ├── CMakeLists.txt
            │   │   ├── rs_driver_pcdsaver.cpp
            │   │   └── rs_driver_viewer.cpp
            │   └── win
            │       ├── demo_online
            │       │   └── demo_online.vcxproj
            │       ├── demo_pcap
            │       │   └── demo_pcap.vcxproj
            │       ├── rs_driver.sln
            │       └── rs_driver_viewer
            │           └── rs_driver_viewer.vcxproj
            ├── source
            │   ├── source_driver.hpp
            │   ├── source.hpp
            │   ├── source_packet_ros.hpp
            │   └── source_pointcloud_ros.hpp
            └── utility
                ├── common.hpp
                └── yaml_reader.hpp

53 directories, 298 files

TZECHIN6, 2023-10-10:
from what i saw in here is that, you put all the packages outside the `src` directory.

If you follow the ros2_ws structure should look like this:
```
ros2_ws
└── src
    ├── package1
    ├── package2
    └── ...
```

Under the root dir `ros2_ws`, run cmd `rosdep install --from-paths src --ignore-src -r -y` should install all required dependencies.

AlessioGianu, 2023-10-11:
Hi, maybe the tree structure shows not correctly the structure; i don't know why but the structure is like you posted and the command for dependencies tell me all is installed correctly. I will try on another computer to understand better why this happens

AlessioGianu, 2023-10-16:
I'm not using any environments at all.
I will try to reinstall all as you suggests, and than the next days there will be a colleague of mine  that  had used this sensor before so maybe with 4 hands we will be able to solve this issue. Thanks for now and if the problems subsist, and if you have time and you want, i will ask you again. 
Thanks so much for your time and your suggestions.

TZECHIN6, 2023-10-16:
Yeah I will reply when I am free. Just leave the finding and step to reproduce in here.