RoboSense-LiDAR/rslidar_sdk #127: Docker

schreiterjp, 2023-07-14:
Hi, 

thanks for the implementation into ROS2, worked like a charm on plain linux once i had the right configuration. (Jetson Nano, Ubuntu 18.04, ROS2 Eloquent) 

Im trying to get it to work inside a docker container running from a compose file. Image is build upon ubuntu 22.04 and ROS2 Humble. 
I guess it it just a network problem regarding container and host connection. Maybe someone already did this and has any advice? 

I will add the docker and compose files later to this issue. 

Thanks.

schreiterjp, 2023-07-26:
Should work on ubuntu using —net=host. 

For Windows WSL2 based docker and MacOS i have no solution.

closed 2023-09-15 by schreiterjp

DrShIkIgAmy, 2023-12-18:
It's the stupidest issue I've ever read. What is the problem, what is the question, what it's all about???? The only question is "Has someone any advice?". Yes god damn it: wash ur teeth every day, drink clean plain water and do sport. HAVE A NICE DAY