RoboSense-LiDAR/rslidar_sdk #131: Feature request: fusion with image

schreiterjp, 2023-09-15:
Hi, 

a nice addition to the driver would be a ROS2 node that subscribes to an image topic and projects the lidar point cloud into the image plane. A projection matrix could possibly be provided with a config.yaml. So the user only has to calibrate the lidar and camera. 

The other way around, projecting an image pixel into the point cloud would give the opportunity to colorise the points and gather additional information. Therefore the inverted matrix could be used. 

Thanks.

AbdullahGM1, 2024-10-09:
here is a package for ros2 that I have created 
https://github.com/AbdullahGM1/ros2_lidar_camera_fusion_with_detection

robbie2021, 2025-02-24:
> here is a package for ros2 that I have created https://github.com/AbdullahGM1/ros2_lidar_camera_fusion_with_detection

This is a python package, is it usable for the c++ SDK?

AbdullahGM1, 2025-02-24:
> > here is a package for ros2 that I have created https://github.com/AbdullahGM1/ros2_lidar_camera_fusion_with_detection
> 
> This is a python package, is it usable for the c++ SDK?

Here is the package in C++

https://github.com/AbdullahGM1/l2i_fusion_detection

robbie2021, 2025-03-04:
> > > here is a package for ros2 that I have created https://github.com/AbdullahGM1/ros2_lidar_camera_fusion_with_detection
> > 
> > 
> > This is a python package, is it usable for the c++ SDK?
> 
> Here is the package in C++
> 
> https://github.com/AbdullahGM1/l2i_fusion_detection

 Thank you for your share.  But i don't understand how this package is used with rslidar_sdk,  can you explain it or make an example?