RoboSense-LiDAR/rslidar_sdk #139: Not receiving any messages with error: ERRCODE_MSOPTIMEOUT

maxpolzin, 2024-04-25:
We have a BPearl Lidar (IP 192.168.1.200) and it is connecting to a freshly installed Ubuntu 22.04 host running ROS2 Humble. Wireshark sees UDP packages on the ports 6699 and 7788. The ROS2 drivers compile just fine, though when starting at a rate of 1Hz this error is shown:

`
ERRCODE_MSOPTIMEOUT 
`

Do you have any ideas how to resolve it? Thank you for your help in advance.

I am attaching screenshots to show the configurations in detail.

[image]
[image]
[image]
[image]
[image]

JTShuai, 2024-05-03:
Do you allow the ports to pass the firewall?

maxpolzin, 2024-05-03:
I receive the packages in Wireshark. If I record a PCAP file in Wireshark,
I can also visualize it in the viewer.

Thus, I conclude the firewall is not the problem, but rather the SDK that
trys to decode/receive the packages.

How should I check if the firewall is nevertheless an issue? Are you
talking about the Linux firewall, ufw?

nawfalz, 2024-05-08:
I had the same issue. I managed to resolve it by running the following commands:
$ sudo ufw allow 6699/udp
$ sudo ufw allow 7788/udp

maxpolzin, 2024-05-08:
@JTShuai and @nawfalz: Thank you for your inputs.

Enabling the ports in the firewall has resolved the issue! 

What was puzzling me to this point was able to receive and record packages with Wireshark. I didn't understand how Wireshark and the Firewall interplayed.

closed 2024-05-08 by maxpolzin

JTShuai, 2024-05-08:
> @JTShuai and @nawfalz: Thank you for your inputs.
> 
> Enabling the ports in the firewall has resolved the issue!
> 
> What was puzzling me to this point was able to receive and record packages with Wireshark. I didn't understand how Wireshark and the Firewall interplayed.

 Wireshark operates "before" firewall and sees all the packets that the network card receives. Then the packets get filtered by the firewall.

For more information, please check the link: https://ask.wireshark.org/question/31184/wireshark-capture-packets-after-or-before-firewall/