RoboSense-LiDAR/rslidar_sdk #146: Colcon build Error [ROS2 Jazzy] [Ubuntu 24.04 Noble]

Rothhammer, 2024-06-13:
When trying to build the sdk in Ros2 Jazzy with colcon I get unexpected errors. All dependencies have been met. The repo, when set up exactly the same with Ros2 Humble in 22.04, compiles without problems.

I am using rs_driver as a submodule in rslidar_sdk.

In Jazzy and 24.04 I get a lot of this:

```
/opt/ros/jazzy/include/rclcpp/rclcpp/any_subscription_callback.hpp:760:27: error: expected primary-expression before ‘,’ token
  760 |           std::is_same_v<T, ConstRefSharedConstPtrSerializedMessageWithInfoCallback>||
      |                           ^
/opt/ros/jazzy/include/rclcpp/rclcpp/any_subscription_callback.hpp:760:84: error: expected primary-expression before ‘>’ token
  760 |           std::is_same_v<T, ConstRefSharedConstPtrSerializedMessageWithInfoCallback>||
      |                                                                                    ^
/opt/ros/jazzy/include/rclcpp/rclcpp/any_subscription_callback.hpp:760:85: error: expected primary-expression before ‘||’ token
  760 |           std::is_same_v<T, ConstRefSharedConstPtrSerializedMessageWithInfoCallback>||
      |                                                                                     ^~
/opt/ros/jazzy/include/rclcpp/rclcpp/any_subscription_callback.hpp:761:16: error: ‘is_same_v’ is not a member of ‘std’; did you mean ‘is_same’?
  761 |           std::is_same_v<T, SharedPtrSerializedMessageCallback>||
      |                ^~~~~~~~~

[...]

  958 |   get_variant()
      |   ^~~~~~~~~~~
/opt/ros/jazzy/include/rclcpp/rclcpp/any_subscription_callback.hpp:964:3: error: no type named ‘variant_type’ in ‘using rclcpp::AnySubscriptionCallback<rslidar_msg::msg::RslidarPacket_<std::allocator<void> >, std::allocator<void> >::HelperT = struct rclcpp::detail::AnySubscriptionCallbackHelper<rslidar_msg::msg::RslidarPacket_<std::allocator<void> >, std::allocator<void>, false, false>’ {aka ‘struct rclcpp::detail::AnySubscriptionCallbackHelper<rslidar_msg::msg::RslidarPacket_<std::allocator<void> >, std::allocator<void>, false, false>’}
  964 |   get_variant() const
      |   ^~~~~~~~~~~
/opt/ros/jazzy/include/rclcpp/rclcpp/any_subscription_callback.hpp:974:34: error: no type named ‘variant_type’ in ‘using rclcpp::AnySubscriptionCallback<rslidar_msg::msg::RslidarPacket_<std::allocator<void> >, std::allocator<void> >::HelperT = struct rclcpp::detail::AnySubscriptionCallbackHelper<rslidar_msg::msg::RslidarPacket_<std::allocator<void> >, std::allocator<void>, false, false>’ {aka ‘struct rclcpp::detail::AnySubscriptionCallbackHelper<rslidar_msg::msg::RslidarPacket_<std::allocator<void> >, std::allocator<void>, false, false>’}
  974 |   typename HelperT::variant_type callback_variant_;
      |                                  ^~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/rslidar_sdk_node.dir/build.make:76: CMakeFiles/rslidar_sdk_node.dir/node/rslidar_sdk_node.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/rslidar_sdk_node.dir/build.make:90: CMakeFiles/rslidar_sdk_node.dir/src/manager/node_manager.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:154: CMakeFiles/rslidar_sdk_node.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< rslidar_sdk [5.25s, exited with code 2]                               

Summary: 2 packages finished [29.4s]
  1 package failed: rslidar_sdk
  1 package had stderr output:  rslidar_sdk
  3 packages not processed

```

JoLichtenfeld, 2024-07-16:
Have you found a solution to this?

Rothhammer, 2024-07-16:
Unfortunately not. I rolled back to Ubuntu 22.04 and it works fine.

JoLichtenfeld, 2024-07-16:
Ok I think it is quite simple. `std::is_same_v` etc. requires C++17, so we need to set in in the CMakeLists.txt accordingly.
The easiest way: Change [this line](https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/ff4517f21f712048b85123a23a58c8cd9e651b92/CMakeLists.txt#L66) to 'jazzy' instead of 'humble':

`if($ENV{ROS_DISTRO} STREQUAL "jazzy")`

This resolves the problem for me running Ubuntu 24.04 and ROS2 jazzy.

Rothhammer, 2024-07-19:
Thank you very much. I can't believe I didn't see that...

jfrascon, 2025-09-13:
Please, close this issue since this is fixed in the CMakeLists.txt, as of 2025-09-13

benmccann, 2026-01-17:
Though the readme still doesn't mention Jazzy as supported, so that should be updated