RoboSense-LiDAR/rslidar_sdk #163: E1R

markalgowis, 2025-01-26:
Hello,
does the E1R lidar is also supported?

HaoQChen, 2025-02-07:
code that support E1R and Airy will push to github soon, you could contant our tech Support for latest version

closed 2025-02-07 by HaoQChen

james-ward, 2025-03-17:
@markalgowis Did you get the E1R working with this SDK? I can see that the E1 is supported in the code, but don't know if that includes the E1R.

markalgowis, 2025-03-17:
> [@markalgowis](https://github.com/markalgowis) Did you get the E1R working with this SDK? I can see that the E1 is supported in the code, but don't know if that includes the E1R.

I'm still waiting for positive replies before buying the sensor...

HaoQChen, 2025-03-18:
hi everyone, E1R and Airy are supported, you can clone the latest version for that

Harin99, 2025-05-14:
Dear @HaoQChen , 

Thank you very much for your kind support with the E1R device. 
I am planning to use the E1R with ROS1 noetic and would like to confirm one thing: 
In the ```config.yaml``` file, should the lidar_type be set to ```RSE1R``` or is ```RSE1``` sufficient? 

Thank you again for your assistance.