RoboSense-LiDAR/rslidar_sdk #166: 连接在线雷达出现ERRCODE_MSOPTIMEOUT

robbie2021, 2025-02-21:
hi,
我的配置信息如下

[image]

运行sdk后报错：

[image]

wireshark能抓取到数据，pcap文件在本地也能正常显示，尝试过关闭防火墙，都无效。

求解！！！

FlinkTheSecond, 2025-09-24:
我的解决办法是在雷达的网页端找到这个setting里的Destination IP Address:把它改成你电脑网口设置的那个ip，然后就可以正常跑通了

[image]

wzhaofeng, 2025-09-24:
> 我的解决办法是在雷达的网页端找到这个setting里的Destination IP Address:把它改成你电脑网口设置的那个ip，然后就可以正常跑通了
> 
> <img alt="Image" width="410" height="246" src="https://private-user-images.githubusercontent.com/176638308/493254545-b2dc9320-236b-4271-8098-cf11ee8b1a26.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg3MDU5NjYsIm5iZiI6MTc1ODcwNTY2NiwicGF0aCI6Ii8xNzY2MzgzMDgvNDkzMjU0NTQ1LWIyZGM5MzIwLTIzNmItNDI3MS04MDk4LWNmMTFlZThiMWEyNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkyNFQwOTIxMDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYmU4YjI1ODZhMWIyNzFlZmNlZTZiNGZmODQxNzM5NzQ3OTA3NDVjMWY3OGI0MDUwNWIwZWIwNjI1ZWMxOTNmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.FVgbsYdVHfkTR8m33HI2rATrZeJXPrJQPxtCYeAYUcA">

我的方法是在修改网址前就用网线连接上雷达，修改好后就断开雷达连接（网线直接断开），然后重新连接就能用了