RoboSense-LiDAR/rslidar_sdk #167: Airy雷达的 IMU坐标系与雷达坐标系的变换关系

Toushine, 2025-02-24:
Airy雷达手册未提供 IMU与LiDAR坐标系的外参

Toushine, 2025-04-17:
> I was looking for the same thing and this is what I figured out. Perhaps it could be helpful for you:
> 
> [#172](https://github.com/RoboSense-LiDAR/rslidar_sdk/issues/172)
Thanks for pointing it out.I print the calibration information in [DecoderRSAIRY](https://github.com/RoboSense-LiDAR/rs_driver/blob/main/src/rs_driver/driver/decoder/decoder_RSAIRY.hpp#L319). 
Finally, I get calibration result as follow,it works normally in LIO system.
IMU frame in LiDAR frame:
q[x,y,z,w]:[ -0.70629, 0.707913, -0.00370051, 0.000629245 ]
t[x,y,z]: [0.00425, 0.00418, -0.00446]

linden713, 2025-06-24:
Hi @Toushine, thanks for sharing the way to print the calibration result. But I couldn't find any LiDAR-IMU extrinsic calibration information around line 319 in decoder_RSAIRY.hpp. It seems that part of the code only handles raw IMU data. Could you kindly point me to the exact location where the extrinsic parameters are handled or printed? Thank you!

Toushine, 2025-06-24:
> Hi [@Toushine](https://github.com/Toushine), thanks for sharing the way to print the calibration result. But I couldn't find any LiDAR-IMU extrinsic calibration information around line 319 in [...]

Sorry, I should use function name which is decodeDifopPkt in decoder_RSAIRY.hpp file. Follow Airy product documentation,it save IMU_CALIB_DATA with DIFOP. In decodeDifopPkt function, device_info_ get the extrinsic calibration information.

Toushine, 2025-06-25:
> Hi [@Toushine](https://github.com/Toushine), thanks! Just to check — is the “Airy product documentation” you mentioned referring to [this user guide](https://www.robosense.ai/en/resources-161)?
> 
> Also, regarding the IMU_CALIB_DATA, I wanted to ask: Is this calibration data something you measured and printed manually, or was it already embedded in the DIFOP packet by the factory?
> 
> I tried printing out the values from decodeDifopPkt() (specifically from device_info_), but they seem to be all zeros. My LIO system runs fine with your printed values, but I’d like to verify whether they were factory-calibrated or user-defined.
> 
> Thanks in advance!

Yes, it defined in C.17 IMU_CALIB_DATA, according to https://github.com/RoboSense-LiDAR/rs_driver/issues/57, it was calibrated individually by RoboSense and saved in DIFOP.
I just print the device_info's x,y,z and qx,qy,qz,qw. 

``` c++
template <typename T_PointCloud>
inline void DecoderRSAIRY<T_PointCloud>::decodeDifopPkt(const uint8_t* packet, size_t size)
{
  const RSAIRYDifopPkt& pkt = *(const RSAIRYDifopPkt*)(packet);
  this->template decodeDifopCommon<RSAIRYDifopPkt>(pkt);

  this->device_info_.qx = convertUint32ToFloat(ntohl(pkt.qx)) ;
  this->device_info_.qy = convertUint32ToFloat(ntohl(pkt.qy)) ;
  this->device_info_.qz = convertUint32ToFloat(ntohl(pkt.qz)) ;
  this->device_info_.qw = convertUint32ToFloat(ntohl(pkt.qw)) ;

  this->device_info_.x = convertUint32ToFloat(ntohl(pkt.x)) ;
  this->device_info_.y = convertUint32ToFloat(ntohl(pkt.y)) ;
  this->device_info_.z = convertUint32ToFloat(ntohl(pkt.z)) ;
  this->device_info_.state = true;

  // Print the imu calibration information
  static bool output = true;
  if(output){
    output = false;
    RS_INFOL << "imu_calib rotation[x,y,z,w]:[" << this->device_info_.qx << ","
             << this->device_info_.qy << "," << this->device_info_.qz << ","
             << this->device_info_.qw << "]" << " translation[x,y,z]:["
             << this->device_info_.x << "," << this->device_info_.y << ","
             << this->device_info_.z << "]" << RS_REND;
  }
}
```

linden713, 2025-06-25:
Hi @Toushine, I found the issue — it was a problem with my print statement.
Now I can see the extrinsic parameters, and they are slightly different from yours:
`q[x,y,z,w]: [-0.706479, 0.707706, 0.00372143, -0.00497837]  
t[x,y,z]  : [0.00425, 0.00418, -0.00446]`
I believe these are from factory calibration. Thanks again!